PilzDE/pilz_robots #53: Gazebo crashes at start up

hslusarek, 2019-01-14:
Up to commit eb203fdbc63503d53c2cfe956dc2a1c0d5fb7470 Gazebo starts without problems. From commit 71bf950fc0a6fdacca01faf0239f43ab1a8eb299 forth Gazebo  immediately crashes at start-up. The only information I get on the terminal is:

`[gazebo_gui-3] process has died [pid 31572, exit code 139, cmd /opt/ros/kinetic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/hslusarek/.ros/log/b9a1d926-17fc-11e9-a60e-1866da344b68/gazebo_gui-3.log].
log file: /home/hslusarek/.ros/log/b9a1d926-17fc-11e9-a60e-1866da344b68/gazebo_gui-3*.log
`
which basically says nothing. However, the problem seems be the mesh file change for link 5.

agutenkunst, 2019-01-15:
I checked with Gazebo version 7.14.0 and e92dc65 and it worked.
Maybe this is a missing <run_depend>?

hslusarek, 2019-01-15:
I have Gazebo version 7.0.0.

agutenkunst, 2019-01-16:
One mesh was broken. New mesh was generated and added in 5f5f65a4ef747d4567b57fc75dd1c92684ab5f3b

closed 2019-01-16 by agutenkunst