PilzDE/pilz_robots #61: Do not depend on meta-package(s)

jschleicher, 2019-01-17:
prbt_moveit_config depends on ros_control and on ros_controllers, but according to [REP-140](http://www.ros.org/reps/rep-0140.html#metapackage) this is not allowed (see discussion in https://github.com/catkin/catkin_tools/issues/370)

For me the current situation causes error outputs of catkin build:
```
Abandoned <<< prbt_gazebo                                   [ Depends on unknown jobs: ros_control ]                                     
Abandoned <<< prbt_moveit_config                            [ Depends on unknown jobs: ros_control ] 
```
and in some (not reproducible) situation @JonathanGruner experienced an error in the first step of `catkin_make` so that the build failed completely.

agutenkunst, 2019-01-17:
Can you post the catkin build configuration?

jschleicher, 2019-01-18:
I can confirm your fix: ff3bd28 doesn't show the "Abondoned" warning any longer.

commit 3b619bab (2019-01-17): Remove exec_depend on metapackages Closes #61
  * Remove ros_control as dep
  * Changed ros_controllers to joint_trajectory_controller
commit a5f53387 (2019-01-17): Remove exec_depend on metapackages Closes #61
  * Cleanup deps
commit ff3bd284 (2019-01-17): Remove exec_depend on metapackages Closes #61
  * Cleanup deps
commit 47f1e896 (2019-01-21): Remove exec_depend on metapackages Closes #61
  * Cleanup deps
closed 2019-01-21 by agutenkunst