PilzDE/pilz_robots #65: Start no possible because of unreasonable joint state

hslusarek, 2019-02-13:
Error:
```
EMCY: 83#0000000000000000
EMCY: 84#0000000000000000
EMCY: 85#0000000000000000
EMCY: 86#0000000000000000
EMCY: 87#0000000000000000
EMCY: 88#0000000000000000
[robot_init-6] process has finished cleanly
log file: /home/jgruner/.ros/log/7961ebfc-e1a7-11e8-b4cf-b7756f5a66b8/robot_init-6*.log
EMCY: 88#008421E4C1E8F5C7
[ERROR] [1541497011.749882349]: Node has emergency error
```

Angle:
```
header: 
  seq: 1094
  stamp: 
    secs: 1541497033
    nsecs: 572782638
  frame_id: ''
name: [prbt_joint_1, prbt_joint_2, prbt_joint_3, prbt_joint_4, prbt_joint_5, prbt_joint_6]
position: [-0.025691246589356526, -0.41233403578366035, 0.7356039198380501, -0.001902408884673819, 1.9980005678055484, 15.406073667231507]
velocity: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
effort: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```

hslusarek, 2019-02-13:
Moved real robot with rviz and suddenly axis stood in axis limit.

hslusarek, 2019-02-13:
Refinement meeting (06.12.2018):
TODO: Test that if robot is moved with Rviz that limits are considered.
TODO: Test that if values outside limits are send over CAN, robot hardware reacts properly.
TODO: Stress test to check if timing error/race condition triggers error.
If both tests show now incorrect behaviour, close bug?

agutenkunst, 2019-02-13:
Could you explain in more detail what you did/expected and what happend?

hslusarek, 2019-02-13:
This bug is a little bit older and I cannot really recall how exactly this bug occurred. Therefore, I suggest to simply check the TODO's from the Refinement and if no problem is detected, to close the bug.

ct2034, 2019-08-30:
@hslusarek What is the state on this?

martiniil, 2019-09-05:
ping @hslusarek

jschleicher, 2020-06-23:
Closing outdated issue. Please reopen with more info, if you experience a similar problem.

closed 2020-06-23 by jschleicher