PilzDE/pilz_robots #72: rosparam can

JonathanGruner, 2019-02-15:
Please insert the hint, that you could change with the param

`<rosparam param="/prbt/driver/bus/device">canx</rosparam>`

with x the number of Can in the moveit_planning_execution.launch.

agutenkunst, 2019-02-21:
Is https://github.com/PilzDE/pilz_robots/blob/3e8b31ea3ebceb120be15aec609a5ae6af55db76/prbt_support/launch/robot.launch#L15-L17

sufficient?

jschleicher, 2019-02-21:
Yep. This issue is about **documentation**.

Maybe the future tutorial about connecting the real robot would be a good place. Maybe we should put this into an example package together with ready-to-use-script(s). I think this should be visible somewhere on wiki.ros.org. What do you think?

commit cac92310 (2019-03-04): update the documentation

* add missing launch file parameters
* rviz_default parameter avoids resetting to the default rviz config at
  every start (or have to clone the repository to make this change)
* reference the launch file template in the tutorials repo. Closes #72.
PR #77 (2019-03-13): update the documentation
* add missing launch file parameters
* rviz_default parameter avoids resetting to the default rviz config at
  every start (or have to clone the repository to make this change)
* reference the launch file template in the tutorials repo. Closes #72.

This adds documentation to the paramters introduced in d0c9fd432c4e475947410940db4667353501d8a9.
commit f2c42b09 (2019-04-03): update the documentation

* add missing launch file parameters
* rviz_default parameter avoids resetting to the default rviz config at
  every start (or have to clone the repository to make this change)
* reference the launch file template in the tutorials repo. Closes #72.
closed 2019-03-13 by agutenkunst