PilzDE/pilz_robots #76: prbt ik fast plugin not found (install space)

ipa-jfh, 2019-02-25:
### Commit
current (3e8b31e)

### Steps to reproduce
```
sudo apt install ros-kinetic-pilz-robots
source /opt/ros/kinetic/setup.bash
roslaunch prbt_moveit_config moveit_planning_execution.launch
```

### Expected behavior
Simulated robot with MoveIt functionality.

### Observed behavior
MoveIt fails and one does e.g. not get the interactive marker.

### Logfiles

```
[ERROR] [1551087039.833746901]: The kinematics plugin (manipulator) failed to load. Error: Failed to load library /opt/ros/kinetic/lib//libprbt_manipulator_moveit_ikfast_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libmoveit_kinematics_base.so.0.9.15: cannot open shared object file: No such file or directory)
[ERROR] [1551087039.833831435]: Kinematics solver could not be instantiated for joint group manipulator.
```
It does work if I build the package `prbt_ikfast_manipulator_plugin` from source. 
I would imagine that an installation step is missing for the `libprbt_manipulator_moveit_ikfast_plugin.so`.

jschleicher, 2019-02-25:
Please try a full `apt upgrade`. Probaly you have an old version of kinematics_base installed and are missing the recent kinematics_base.so?

From your error message:
```
libmoveit_kinematics_base.so.0.9.15
```

If you install new packages you should always `apt update && apt upgrade` first. (SONAMES and ABI compatibility...)

ipa-jfh, 2019-02-25:
Thank you for your quick answer. You are right, the upgrade solved the issue.

closed 2019-02-25 by ipa-jfh