PilzDE/pilz_robots #80: No modbus sto

JonathanGruner, 2019-03-01:
I have a problem during modbus connection.

I connected the robot over a PSS4000 with my computer and tested the connection with @martiniil with pymodbus, all required datas could be read in the right registers. 

But when starting the robot there comes "Modbus Disconnected".
If I then release the enabling switch, the terminal says "Node has emergency Error".

Furthermore the robot doesen't want to move at any time.

You can find the project on "ros-workshop" in gitlab if you want to look into it.

JonathanGruner, 2019-03-05:
The problem couldn't be solved yet with sevaral computers , same same settings than the PLC in the ROS-laboratory, also other PLCs...

jschleicher, 2019-03-05:
Hallo Jonathan,
thanks for your report. An PSS4000 setup that you are trying is not officially supported. We will check, if this is a viable configuration in the future. The current ROS packages are supported with "pnoz" only.

closed 2019-03-05 by jschleicher