PilzDE/pilz_robots #92: Robot cannot be moved after Emergency-Stop/Panic button/ Enabling button release

hslusarek, 2019-03-26:
### Commit
- [ ] @hslusarek TODO Add commit.

### Steps to reproduce
Start real robot via:  
```roslaunch prbt_moveit_config moveit_planning_execution.launch sim:=false pipeline:=pilz_command_planner```
1. Perform motion (e.g. vial Python script)
2. During motion do one of the following things:
    - release Enabling button,
    - bush panic button or
    - push emergency button.
3. If no problem occures go back to step 1.

### Expected behavior
After I do one of the following things it should be possible to continue to move the robot:  
- release Enabling button,
- bush panic button or
- push emergency button.

### Observed behavior
It is not possible to continue to move the robot because on the console it is stated that:
```PATH_TOLERANCE_VIOLATED```

hslusarek, 2019-03-28:
- [ ] TODO: Reproduce bug at robot in lab.

jschleicher, 2019-04-16:
Maybe duplicate of #93?

hslusarek, 2019-04-26:
- [ ] TODO: Determine Firmeware version of robot on which bug occured

hslusarek, 2019-05-17:
Bug was reproduced with robot with new firmeware and kinetic-devel

## Version details
- pilz_industrial_motion: a921718f1eaf7dcca57c61bce28f3e0c53c79015
- pilz_robots: a0d4fda5df623de8477f8acf6affaf9bb900716f
- prbt_grippers: 31f9f169f82839512fc4c376104e6bd09741a1cf
- Firmeware: 101 Build:11345 Date:2018-12-17 07:33:065

## Console output
```
SUMMARY
========

CLEAR PARAMETERS
 * /prbt/driver/

PARAMETERS
 * /controller_joint_names: ['prbt_joint_1', ...
 * /move_group/allow_trajectory_execution: True
 * /move_group/capabilities: pilz_trajectory_g...
 * /move_group/controller_list: [{'action_ns': 'f...
 * /move_group/disable_capabilities: 
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planning_plugin: pilz::CommandPlanner
 * /move_group/planning_scene_monitor/publish_geometry_updates: True
 * /move_group/planning_scene_monitor/publish_planning_scene: True
 * /move_group/planning_scene_monitor/publish_state_updates: True
 * /move_group/planning_scene_monitor/publish_transforms_updates: True
 * /move_group/request_adapters: 
 * /move_group/start_state_max_bounds_error: 0.1
 * /move_group/trajectory_execution/allowed_execution_duration_scaling: 1.2
 * /move_group/trajectory_execution/allowed_goal_duration_margin: 0.5
 * /move_group/trajectory_execution/allowed_start_tolerance: 0.01
 * /pilz_modbus_read_client_node/index_of_first_register_to_read: 512
 * /pilz_modbus_read_client_node/modbus_server_ip: 169.254.60.1
 * /pilz_modbus_read_client_node/modbus_server_port: 502
 * /pilz_modbus_read_client_node/num_registers_to_read: 2
 * /prbt/driver/bus/device: can0
 * /prbt/driver/defaults/eds_file: config/prbt_0_1.dcf
 * /prbt/driver/defaults/eds_pkg: prbt_support
 * /prbt/driver/name: manipulator
 * /prbt/driver/nodes/prbt_joint_1/id: 3
 * /prbt/driver/nodes/prbt_joint_2/id: 4
 * /prbt/driver/nodes/prbt_joint_3/id: 5
 * /prbt/driver/nodes/prbt_joint_4/id: 6
 * /prbt/driver/nodes/prbt_joint_5/id: 7
 * /prbt/driver/nodes/prbt_joint_6/id: 8
 * /prbt/driver/sync/interval_ms: 10
 * /prbt/driver/sync/overflow: 0
 * /prbt/joint_names: ['prbt_joint_1', ...
 * /prbt/manipulator_joint_state_controller/publish_rate: 50
 * /prbt/manipulator_joint_state_controller/type: joint_state_contr...
 * /prbt/manipulator_joint_trajectory_controller/action_monitor_rate: 10
 * /prbt/manipulator_joint_trajectory_controller/constraints/goal_time: 0.6
 * /prbt/manipulator_joint_trajectory_controller/constraints/prbt_joint_1/goal: 0.01
 * /prbt/manipulator_joint_trajectory_controller/constraints/prbt_joint_1/trajectory: 0.157
 * /prbt/manipulator_joint_trajectory_controller/constraints/prbt_joint_2/goal: 0.01
 * /prbt/manipulator_joint_trajectory_controller/constraints/prbt_joint_2/trajectory: 0.157
 * /prbt/manipulator_joint_trajectory_controller/constraints/prbt_joint_3/goal: 0.01
 * /prbt/manipulator_joint_trajectory_controller/constraints/prbt_joint_3/trajectory: 0.157
 * /prbt/manipulator_joint_trajectory_controller/constraints/prbt_joint_4/goal: 0.01
 * /prbt/manipulator_joint_trajectory_controller/constraints/prbt_joint_4/trajectory: 0.157
 * /prbt/manipulator_joint_trajectory_controller/constraints/prbt_joint_5/goal: 0.01
 * /prbt/manipulator_joint_trajectory_controller/constraints/prbt_joint_5/trajectory: 0.157
 * /prbt/manipulator_joint_trajectory_controller/constraints/prbt_joint_6/goal: 0.01
[... 176 lines cut ...]
[ INFO] [1558079444.592362998]: Generating PTP trajectory...
[ INFO] [1558079445.611700625]: Performing STOP due to STO signal
[ WARN] [1558079445.613169774]: Controller handle prbt/manipulator_joint_trajectory_controller/ reports status PREEMPTED
[ INFO] [1558079445.613211361]: Completed trajectory execution with status PREEMPTED ...
[ INFO] [1558079445.814444598]: Halting down XXX
[ INFO] [1558079445.824676851]: ABORTED: Solution found but controller failed during execution
EMCY: 87#0010812903000000
EMCY: 88#0010812903000000
EMCY: 84#0010812903000000
EMCY: 85#0010812903000000
EMCY: 86#0010812903000000
EMCY: 83#0010812903000000
[ INFO] [1558079452.746972692]: Recovering XXX
ER: 129
[ INFO] [1558079456.071845209]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1558079456.071918379]: Planning attempt 1 of at most 1
[ INFO] [1558079456.072009962]: Initialized Point-to-Point Trajectory Generator.
[ INFO] [1558079456.072079169]: Generating PTP trajectory...
[ WARN] [1558079456.885477023]: Controller prbt/manipulator_joint_trajectory_controller/ failed with error code PATH_TOLERANCE_VIOLATED
[ WARN] [1558079456.885571385]: Controller handle prbt/manipulator_joint_trajectory_controller/ reports status ABORTED
[ INFO] [1558079456.885591422]: Completed trajectory execution with status ABORTED ...
[ INFO] [1558079457.088095546]: ABORTED: Solution found but controller failed during execution
[ INFO] [1558079462.595499927]: Performing STOP due to STO signal
[ INFO] [1558079462.798405261]: Halting down XXX
```

hslusarek, 2019-05-17:
Bug also appears with melodic-devel (17.05.2019)

hslusarek, 2019-05-17:
The following workaround seems to fix the bug: [see here](https://github.com/ct2034/ros_canopen/tree/d795a68559009a2653fc5a29c6345a405f1563f4)

hslusarek, 2019-05-17:
> Maybe duplicate of #93?

It seems that way yes.

hslusarek, 2019-06-06:
Close because we assume that this bug is a duplicate #93.

closed 2019-06-06 by hslusarek