PilzDE/pilz_robots #93: Mysterios CAN Error (ER: XY)

JonathanGruner, 2019-03-27:
### Commit
The last few commits

### Steps to reproduce
How can the issue/bug be reproduced?
1. Start the robot with holding the Enabling switch
2. Release the switch, press it again, release it, ..... 
3. The Error will occur and the brakes won't open again after a few (sometimes even after the first) releases of the switch
4. If I don't release the button (or in my case close and lock the door of the cell in automatic mode) the error won't occur as often as with the other tactic.

### Expected behavior
I expected, that the robot will continue working after the enabling switch is pressed again.

### Observed behavior
After Error I have to reboot the program (not the robot) to make it work again

### Logfiles
Will be appended in a few minutes

jschleicher, 2019-03-27:
The emergency `EMCY: 87#001081 29 03000000` is         `ERROR_SAFE_TORQUE_OFF  //0x29`

ct2034, 2019-03-28:
This was **not** reproducible with firmware
```
"100 Build:11158 Date:2018-06-07 16:49:55%"
```

JonathanGruner, 2019-04-01:
I have firmware:
``` "101 Build 11345 Date:2018-12-17 07:33:065" ```
Its different!

ct2034, 2019-05-17:
In the archive there are three can dumps for comparison:
[can_dumps_wireshark.zip](https://github.com/PilzDE/pilz_robots/files/3191531/can_dumps_wireshark.zip)
- `fw100_error129_no_failure_gz.pcapng`: old firmware - error does not appear

martiniil, 2019-09-05:
Closed via #175

closed 2019-09-05 by martiniil