PilzDE/pilz_robots #98: Missing interactive marker in rviz, when using gripper:=pg70

rfeistenauer, 2019-04-05:
### Commit
commit 5913a9009b8aef642b01c9000c80d1227d9a0643
and 
commit 393358665f2c9adf06d9ca5748864dc77ee2d867

(from source)

### Steps to reproduce
$ roslaunch prbt_moveit_config moveit_planning_execution.launch gripper:=pg70

### Expected behavior
interactive marker should be at tcp position

### Observed behavior
no interactive marker available

### Logfiles
No difference to log of starting it without the gripper. But without it I get the marker

### Tried so far
Restarting roscore does not help.

I checked into other issues stating the same problem.
https://answers.ros.org/question/115641/moveit-interactive-markers-missing/ (2014) 
they state labeling the end effektor or fixing parent group  may help, but that was no problem so far

in https://github.com/ros-planning/moveit/issues/736 (2018)
this was pointed to world missing, which is not the case, but the referenced issue is all about
the change in quaternion normalization and initialization issue from the start of the year

maybe the warning:
`
[ WARN] [1554452220.703757035]: Interactive marker 'EE:goal_prbt_flange' contains unnormalized quaternions. This warning will only be output once but may be true for others; enable DEBUG messages for ros.rviz.quaternions to see more details.
`
has something to do with it, but it also appears when launching without the gripper ...

jschleicher, 2019-04-05:
What version of `prbt_grippers` are you using? I suspect https://github.com/PilzDE/prbt_grippers/commit/ea840fafba1d2bbd523e42d3e1c567d48234c7ad could be the cause?

rfeistenauer, 2019-04-05:
ros-kinetic-prbt-grippers is already the newest version (0.0.2-0xenial-20190401-162607-0800)

rfeistenauer, 2019-04-05:
meaning yes it could, but no it doesn't seem to be the problem.
build and run ea840fa => did not solve the issue