PilzDE/pilz_robots #100: Calling update_ikfast by catkin

ct2034, 2019-04-12:
In the spirit of only checking in source files, it would make sense to *not* check in the auto-generated ikfast cpp files. 

Instead, it would be more streamlined to call them out of catkin: https://stackoverflow.com/questions/39960173/run-custom-shell-script-with-cmake

This would also future-proof this for when something changes in moveit.

jschleicher, 2019-04-15:
I don't think this is feasible for the IK plugin. We have several manual acceptance tests, that (depending on the diff) must be executed. A regression due to an automated re-generation would not be detected.

If you're talkin about the solver itself: That is actually patched manually (tweak) and must not be re-generated without thouroughly testing it and running the benchmarks (which aren't automated either).

ct2034, 2019-04-25:
okay, maybe this is not a good idea, then.

closed 2019-04-25 by ct2034

jschleicher, 2019-04-29:
It IS a good idea, but currently not feasible :-(