PilzDE/pilz_robots #118: Pilz_tutorial_4 with real robot

OndrejNedelka, 2019-05-06:
### Commit
At which commit was the issue/bug observed?

### Steps to reproduce
How can the issue/bug be reproduced?
`roslaunch pilz_tutorial my_application.launch`

### Expected behavior
Physical robot should move as in RVIZ.

### Observed behavior
Launch fails at error:
`unused args [rviz_config] for include of [/opt/ros/kinetic/share/prbt_moveit_config/launch/moveit_rviz.launch]
The traceback for the exception was written to the log file`

My package:
(Tutorial nr. 4 with modified launch file)
[Uploading pilz_tutorial_4.tar.gz…]()

Version of used packages
[DPKG_pilz.txt](https://github.com/PilzDE/pilz_robots/files/3148239/DPKG_pilz.txt)
[DPKG_prbt.txt](https://github.com/PilzDE/pilz_robots/files/3148241/DPKG_prbt.txt)

Terminal output:
[roslaunch output.docx](https://github.com/PilzDE/pilz_robots/files/3148243/roslaunch.output.docx)

jschleicher, 2019-05-06:
The attachment link of your launch file does not work. Can you please share your (modified) launch file?

OndrejNedelka, 2019-05-06:
> The attachment link of your launch file does not work. Can you please share your (modified) launch file?

It should work now.
Thank you.

martiniil, 2019-05-07:
The `rviz_config` argument is added in `prbt_moveit_config 0.4.8`. So it will be out only with the next kinetic release.

closed 2019-05-07 by martiniil