PilzDE/pilz_robots #123: prbt_ikfast_manipulator_plugin fails to compile on armhf

jschleicher, 2019-05-09:
### Commit
At which commit was the issue/bug observed?

### Steps to reproduce
```

### Expected behavior
Describe what should happen

### Observed behavior
Describe what happens

### Logfiles
If there are logfiles, append them.

### Reference to test
If there is no test that can detect the issue/bug, please create one and
reference it here.

agutenkunst, 2019-05-16:
Could we use a flag for armhf to enable c++17?

SansoneG, 2019-05-17:
A little more system information to @jschleicher's comment:
```
$ lscpu
Architecture:        armv7l
Byte Order:          Little Endian
CPU(s):              4
On-line CPU(s) list: 0-3
Thread(s) per core:  1
Core(s) per socket:  4
Socket(s):           1
Vendor ID:           ARM
Model:               4
Model name:          Cortex-A53
Stepping:            r0p4
CPU max MHz:         1200.0000
CPU min MHz:         600.0000
BogoMIPS:            38.40
Flags:               half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
```
Linux Kernel Version: `4.9.76-rt60-v7+`

SansoneG, 2019-05-20:
Setting up `prbt_ikfast_manipulator_plugin/CMakeLists.txt` like this also fixes the issue.

```
add_compile_options(-Wall)
add_compile_options(-Wextra)
add_compile_options(-Wno-unused-parameter)
add_compile_options(-Wno-unused-variable)
add_compile_options(-Werror)

if(CMAKE_COMPILER_IS_GNUCXX) 

if (CMAKE_CXX_COMPILER_VERSION VERSION_GREATER 7.0) 
set(CMAKE_CXX_FLAGS

"${CMAKE_CXX_FLAGS} -faligned-new")

endif() 
endif() 
```

jschleicher, 2019-05-20:
@SansoneG thanks for testing.
I've opened that as pull-request.

PR #129 (2019-06-11): enable aligned new
when compiling with gcc7+

Fixes #123
commit 3945de56 (2019-06-11): enable aligned new

when compiling with gcc7+

Fixes #123
commit 84494d08 (2019-06-17): enable aligned new

when compiling with gcc7+

Fixes #123

(cherry picked from commit 3945de563228c072ce18b0b8a73eb4b66299df5e)
commit 8f95964a (2019-06-17): enable aligned new

when compiling with gcc7+

Fixes #123

(cherry picked from commit 3945de563228c072ce18b0b8a73eb4b66299df5e)
closed 2019-06-11 by agutenkunst