PilzDE/pilz_robots #128: max time [60.0s] allotted for test [integrationtest_brake_test_required] of type [prbt_hardware_support/integrationtest_brake_test_required]

ct2034, 2019-05-16:
### Commit
https://github.com/PilzDE/pilz_robots/tree/d653264b47adb1c2fbe50e271d21ba8d1c625b8d

### Steps to reproduce
How can the issue/bug be reproduced?
1. Build Tests
2. `rostest prbt_hardware_support integrationtest_brake_test_required.test`

### Expected behavior
Test exits successfully

### Observed behavior
Test times out

### Logfiles
```
[ERROR] [1557999024.967094805]: Could not establish modbus connection.Connection refused
[ERROR] [1557999024.967725786]: Init failed
[ERROR] [1557999024.967743168]: Connection to 127.0.0.1:20701 failed. Try(1/10)
[ERROR] [1557999024.994180516]: Waiting for Node /pilz_modbus_read_client_node
[ERROR] [1557999024.994943383]: Notify
[ERROR] [1557999024.994962549]: Inside Loop
[ERROR] [1557999024.995308584]: Node /pilz_modbus_read_client_node found
[ERROR] [1557999024.995327362]: Waiting for Node /modbus_brake_test_announcer_node
[ERROR] [1557999024.995728553]: Node /modbus_brake_test_announcer_node found
[ERROR] [1557999025.968288826]: modbus_tcp_accept
ERROR Connection reset by peer: read
[Testcase: testintegrationtest_brake_test_required] ... ERROR!
ERROR: max time [60.0s] allotted for test [integrationtest_brake_test_required] of type [prbt_hardware_support/integrationtest_brake_test_required]
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rostest/runner.py", line 148, in fn
    self.test_parent.run_test(test)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rostest/rostest_parent.py", line 132, in run_test
    return self.runner.run_test(test)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/launch.py", line 689, in run_test
    (test.time_limit, test.test_name, test.package, test.type))
--------------------------------------------------------------------------------

[ROSTEST]-----------------------------------------------------------------------

```

### Reference to test
`rostest prbt_hardware_support integrationtest_brake_test_required.test`

martiniil, 2019-05-22:
I cannot reproduce this bug. On my computer the test ran in 3.6s @ct2034

martiniil, 2019-05-23:
@martiniil Check ROS buildfarm
maybe fixed by 5754d7756f0b64130832b132fa4c8031c2e2f315

ct2034, 2019-06-03:
I checked it again also with the prerelease tests. It seems to be fixed by #131. If the problem occurs again - reopen

closed 2019-06-03 by ct2034