PilzDE/pilz_robots #143: Compiling not possible

padhupradheep, 2019-06-13:
```
/home/neobotix/ros_ws_pilz/src/pilz_robots/prbt_hardware_support/src/libmodbus_client.cpp: In member function ‘virtual void prbt_hardware_support::LibModbusClient::setResponseTimeoutInMs(unsigned int)’:
/home/neobotix/ros_ws_pilz/src/pilz_robots/prbt_hardware_support/src/libmodbus_client.cpp:53:51: error: invalid conversion from ‘timeval*’ to ‘uint32_t {aka unsigned int}’ [-fpermissive]
   modbus_set_response_timeout(modbus_connection_, &response_timeout);
                                                   ^~~~~~~~~~~~~~~~~
/home/neobotix/ros_ws_pilz/src/pilz_robots/prbt_hardware_support/src/libmodbus_client.cpp:53:68: error: too few arguments to function ‘int modbus_set_response_timeout(modbus_t*, uint32_t, uint32_t)’
   modbus_set_response_timeout(modbus_connection_, &response_timeout);
                                                                    ^
In file included from /home/neobotix/ros_ws_pilz/src/pilz_robots/prbt_hardware_support/include/prbt_hardware_support/libmodbus_client.h:21:0,
                 from /home/neobotix/ros_ws_pilz/src/pilz_robots/prbt_hardware_support/src/libmodbus_client.cpp:23:
/usr/local/include/modbus/modbus.h:186:16: note: declared here
 MODBUS_API int modbus_set_response_timeout(modbus_t *ctx, uint32_t to_sec, uint32_t to_usec);
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/neobotix/ros_ws_pilz/src/pilz_robots/prbt_hardware_support/src/libmodbus_client.cpp: In member function ‘virtual unsigned int prbt_hardware_support::LibModbusClient::getResponseTimeoutInMs()’:
/home/neobotix/ros_ws_pilz/src/pilz_robots/prbt_hardware_support/src/libmodbus_client.cpp:59:68: error: cannot convert ‘timeval*’ to ‘uint32_t* {aka unsigned int*}’ for argument ‘2’ to ‘int modbus_get_response_timeout(modbus_t*, uint32_t*, uint32_t*)’
   modbus_get_response_timeout(modbus_connection_, &response_timeout);
                                                                    ^

```

Tried to do the Catkin_make, but had this error during compilation! 

Thanks ! 

Pradheep

jschleicher, 2019-06-13:
What is your system?
Ubuntu? ROS-distro?
The tutorials list [requirements](https://wiki.ros.org/pilz_robots/Tutorials/ModelYourApplicationWithPRBT#Prerequisites) where the packages are successfully tested.

Especially we use libmodbus version 3.0.6. Upstream libmodbus version 3.1.2 introduces an [API change](https://libmodbus.org/2015/libmodbus-v3-1-2/) that is not yet contained in the official ubuntu packages and thus not covered by pilz_robots, neither.

jschleicher, 2019-07-02:
Closing as no system information available. Please reopen with additional information.

closed 2019-07-02 by jschleicher