PilzDE/pilz_robots #180: /prbt/joint_states does not publish the velocity and effort values

ipa-hsd-rr, 2019-07-11:
### Commit
8b529fe

### Steps to reproduce
rostopic echo /prbt/joint_states 

### Expected behavior
The /prbt/joint_states should publish the position, velocity and effort values of the robot

### Observed behavior
Only the position values are published. The velocity and effort values remains as zero.

### Logfiles
```
---
header: 
  seq: 4641916
  stamp: 
    secs: 1562851379
    nsecs: 611657848
  frame_id: ''
name: [prbt_joint_1, prbt_joint_2, prbt_joint_3, prbt_joint_4, prbt_joint_5, prbt_joint_6]
position: [1.5610224829837283, -0.08478809506188453, 1.1859512267301469, 0.03310889591033243, 0.29941123317962726, -0.8170235294435855]
velocity: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
effort: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
```

jschleicher, 2019-07-11:
Thanks for your report.

Are you using a simulated or real robot? Please tell us the launch command you use.

If I use gazebo the joint_states are filled completely: `roslaunch prbt_gazebo sim_prbt_gazebo.launch`.

ipa-hsd-rr, 2019-07-11:
I am using it on real robot (Pilz prbt) and I am working on [](https://github.com/rosin-project/automatica18_scan_and_plan_demo)
The launch commands I used was 
```
roslaunch snp_prbt_bringup hardware_bringup.launch
roslaunch snp_prbt_bringup prbt_godel.launch sim_robot:=false 
```

ct2034, 2019-08-30:
Hey Ragesh. Thanks for the report. We are currently very involved in other topics but I want to let you know that we did not forget about this issue. Please let me know if this is currently blocking you in any way which would help us with prioritization.

jschleicher, 2019-10-22:
Working with the real robot, this can be configured with `ros_canopen`. If you add PDO-mapping of velocity registers (see dcf), the velocity actual values are published (same for effort).

Of course this is a performance trade-off: Either you can send set-positions faster (e.g. 5ms cycle-time) or you can read all the actual values back from the robot. I think, the default is okay thus removing the 'bug' label. The templates show, how to [override single paramters](https://github.com/PilzDE/pilz_templates/blob/28094f760e4ba2b95b88500d76a7adc880c63010/prbt_default_application/launch/my_application.launch#L80-L84) and I'd suggest to add a section there how to configure velocity PDO-mapping.

closed 2019-12-03 by ipa-rar