PilzDE/pilz_robots #187: Fix unstable unittest_pilz_joint_trajectory_controller

agutenkunst, 2019-07-22:
### Commit
https://github.com/PilzDE/pilz_robots/commit/69001be9805098286fa23096b3c9d05783b20b67

See https://travis-ci.org/PilzDE/pilz_robots/jobs/562070480

martiniil, 2019-08-08:
You don't have a log by chance?
This could be related to the instability in #206, which seems to vanish or at least was heavily reduced through fd509a8d90da7f2bba1d43aa034da0267f9a788d.

agutenkunst, 2019-10-16:
Did not occurred again so far. Will close for now. Please reopen if needed.

agutenkunst, 2019-10-17:
Reopen ;-)
I just realized you could use sim time in this test.
See http://docs.ros.org/melodic/api/rostime/html/classros_1_1Time.html#ab569bffab97380d8a73275cdc02afb8f this would maybe even enable turning this into a unittest... 

@martiniil what do you think?

leunMar, 2019-10-18:
I would like this very much. But I don't agree on re-opening this issue. Or did you observe that the test is still unstable? If not, I would suggest that you either change the title of this issue (and remove the bug-label) or open a new issue.

leunMar, 2019-10-22:
Test failed again: (exceeds max time 60.0s) https://api.travis-ci.org/v3/job/601221025/log.txt

martiniil, 2020-02-06:
Should be fixed via fcc9c2c.

closed 2020-02-06 by martiniil