PilzDE/pilz_robots #206: Sporadic failure of integrationtest_stop1

martiniil, 2019-08-07:
### Commit
587d05ba2e8fce0fc241e789de48b53dcdd7b9ef

### Steps to reproduce
Rerun travis job, eventually it will fail.
On my local machine I could produce a similar test result as follows:
1. stress -c 60 -m 24 -t 60s
2. rostest prbt_hardware_support integrationtest_stop1.test --text

### Expected behavior
The test should not fail.

### Observed behavior
The test integrationtest_stop1.test fails due to the error 

> Failed to read modbus registers! Connection timed out

from `pilz_modbus_client.cpp`. This could also be the cause of the sporadic failures on travis.

### Logfiles
https://travis-ci.org/PilzDE/pilz_robots/jobs/568226002

martiniil, 2019-08-07:
I assume that fd509a8d90da7f2bba1d43aa034da0267f9a788d fixes it. At least I can no longer produce a failure on the local machine.

jschleicher, 2019-08-07:
Did you ever consider a topic publish/subscribe connection instead of polling a service?

I cannot find any rationale in https://github.com/PilzDE/pilz_robots/pull/139 concerning the architectural decision and from my point of view a publisher would have been the better choice. Not 100% sure, if it would dissolve this test problems but on the first glance it looks so.

martiniil, 2019-08-07:
The rationale is that in the context of the brake test execution we would like to be able to query if the controller is executing.

martiniil, 2019-08-14:
Looks like 1e323584da342d2e0a421fde5c6f3c6a36dedcca resolves the problem. Closing for now.

closed 2019-08-14 by martiniil