PilzDE/pilz_robots #214: Failure of unittest_update_filter

martiniil, 2019-08-14:
### Commit
1e323584da342d2e0a421fde5c6f3c6a36dedcca

### Steps to reproduce
No idea

### Expected behavior
Test should pass

### Observed behavior
Test fails

### Logfiles
https://api.travis-ci.org/v3/job/571708638/log.txt

### Reference to test
prbt_hardware_support/test/unittests/unittest_update_filter.test

martiniil, 2019-08-14:
Log:

```
Full test results for 'build/prbt_hardware_support/test_results/prbt_hardware_support/rostest-test_unittests_unittest_update_filter.xml'

-------------------------------------------------

<?xml version="1.0" encoding="utf-8"?>

<testsuite errors="0" failures="1" name="unittest.suite.TestSuite" tests="1" time="1.247"><testcase classname="rostest.runner.RosTest" name="testunittest_filter_pipeline" time="1.2468"><failure type="AssertionError">test [unittest_filter_pipeline] did not generate test results

  File "/usr/lib/python2.7/unittest/case.py", line 329, in run

    testMethod()

  File "/opt/ros/melodic/lib/python2.7/dist-packages/rostest/runner.py", line 164, in fn

    self.assert_(os.path.isfile(test_file), "test [%s] did not generate test results"%test_name)

  File "/usr/lib/python2.7/unittest/case.py", line 422, in assertTrue

    raise self.failureException(msg)

</failure></testcase><system-out>&lt;![CDATA[

[ROSTEST]setup[/root/catkin_ws/src/pilz_robots/prbt_hardware_support/test/unittests/unittest_update_filter.test] run_id[059fb17c-be97-11e9-bc8f-0242ac110002] starting

&amp;#x001B;[1mstarted roslaunch server http://a3f2136bc90c:38586/&amp;#x001B;[0m

SUMMARY

========

PARAMETERS

 * /rosdistro: melodic

 * /rosversion: 1.14.3

NODES

[ROSTEST]setup[/root/catkin_ws/src/pilz_robots/prbt_hardware_support/test/unittests/unittest_update_filter.test] run_id[059fb17c-be97-11e9-bc8f-0242ac110002] done

[ROSTEST]Running test [unittest_filter_pipeline]

&amp;#x001B;[1mROS_MASTER_URI=http://a3f2136bc90c:38510/&amp;#x001B;[0m

&amp;#x001B;[1msetting /run_id to 059fb17c-be97-11e9-bc8f-0242ac110002&amp;#x001B;[0m

[ROSTEST]running test unittest_filter_pipeline

&amp;#x001B;[1mprocess[unittest_filter_pipeline-1]: started with pid [18066]&amp;#x001B;[0m

&amp;#x001B;[1m[unittest_filter_pipeline-1] process has finished cleanly

log file: 059fb17c-be97-11e9-bc8f-0242ac110002/unittest_filter_pipeline-1*.log&amp;#x001B;[0m

[ROSTEST]test [unittest_filter_pipeline] finished

[ROSTEST]tearDown[/root/catkin_ws/src/pilz_robots/prbt_hardware_support/test/unittests/unittest_update_filter.test]

shutting down processing monitor...

... shutting down processing monitor complete

[ROSTEST]rostest teardown /root/catkin_ws/src/pilz_robots/prbt_hardware_support/test/unittests/unittest_update_filter.test complete

]]&gt;</system-out><system-err>&lt;![CDATA[

]]&gt;</system-err></testsuite>

-------------------------------------------------

build/prbt_hardware_support/test_results/prbt_hardware_support/rostest-test_unittests_unittest_update_filter.xml: 1 tests, 0 errors, 1 failures, 0 skipped

Summary: 148 tests, 0 errors, 1 failures, 0 skipped

/root/ici/tests/source_tests.sh: line 246: error: command not found

ici_time:end:1b47be57:start=1565789131182277648,finish=1565789191891545225,duration=60709267577

ici_fold:end:catkin_run_tests[31m<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<[0m

[31mFunction 'catkin_run_tests' returned with code '127' after 1 min 0 sec[0m

travis_time:end:250871ac:start=1565786700626998264,finish=1565789193748647649,duration=2493121649385
[0K[31;1mThe command ".industrial_ci/travis.sh" exited with 127.[0m

Done. Your build exited with 1.

```

martiniil, 2019-08-14:
PR #173 was about the same test. Unfortunately the issue is not fully resolved.

jschleicher, 2020-06-23:
Closing outdated issue.

closed 2020-06-23 by jschleicher