PilzDE/pilz_robots #228: Please comment your code

JonathanGruner, 2019-08-29:
### Commit
At all of them

### Steps to reproduce
1. Open the files
2. Try to understand the code
3. Search comments--->There you will find the issue

### Expected behavior
Well documented files, especally launch files, config files, etc.

### Observed behavior
Describe what happens

### Logfiles
If there are logfiles, append them.

### Reference to test
If there is no test that can detect the issue/bug, please create one and
reference it here.

ct2034, 2019-08-30:
@JonathanGruner Thanks for your feedback. 
Unfortunately, this is a relatively broad topic. Could you please give an example of where you think we lack documentation. _Also, feel free to open a pull request!_

jschleicher, 2019-09-10:
This concerns e.g. launch files in the folder  

https://github.com/PilzDE/pilz_robots/blob/4d49235f22a0816058a40117aeb5f22b850d31f1/prbt_hardware_support/launch/stop1_executor_node.launch#L18-L20

Sometimes a reference to the main readme would be enough, but in this case the readme doesn't mention `stop1_executor_node` at all.

JonathanGruner, 2019-09-11:
Another great exampel would be:
https://github.com/PilzDE/pilz_robots/blob/4d49235f22a0816058a40117aeb5f22b850d31f1/prbt_hardware_support/launch/modbus_client.launch#L29-L40

I don't want to bother you with a detailed description of each command/parameter, but if you could briefly explain what you are doing I would be happy :)

For example: num_first_register_to_read: First register adress of 4x registers type in modicon notation
num_registers: first = runpermitted, second = baketest, third = sth other

JonathanGruner, 2019-09-18:
https://github.com/PilzDE/pilz_robots/blob/24a1c5dd8ad6be342b27bfef5a3bb13b471003ef/prbt_hardware_support/config/modbus_read_api_spec_pss4000.yaml#L1-L4

Here you could write an comment, that users are able to add different Registers

JonathanGruner, 2020-02-14:
Closed regarding to closed issue #309 .

closed 2020-02-14 by JonathanGruner