PilzDE/pilz_robots #230: Frequency testing in `unittest_pilz_modbus_client` causes sporadic test failures

ct2034, 2019-08-29:
one instability remains .. (https://travis-ci.org/PilzDE/pilz_robots/jobs/578411138#L5011)
```
<failure message="/root/catkin_ws/src/pilz_robots/prbt_hardware_support/test/unittests/unittest_pilz_modbus_client.cpp:518&#x0A;      Expected: expected_value&#x0A;      Which is: 3&#x0A;To be equal to: curr_value&#x0A;      Which is: 4&#x0A;Frequency used by PilzModbusClient is probably too high" type=""><![CDATA[/root/catkin_ws/src/pilz_robots/prbt_hardware_support/test/unittests/unittest_pilz_modbus_client.cpp:518
      Expected: expected_value
      Which is: 3
To be equal to: curr_value
      Which is: 4
Frequency used by PilzModbusClient is probably too high]]></failure>
    </testcase>
```

_Originally posted by @ct2034 in https://github.com/PilzDE/pilz_robots/pull/229#issuecomment-526251507_

martiniil, 2019-09-11:
catched on the ROS buildfarm:
http://build.ros.org/job/Mdev__pilz_robots__ubuntu_bionic_amd64/67/testReport/junit/(root)/PilzModbusClientTests/testSettingReadFrequency/

martiniil, 2019-10-07:
Fixed via #243. Re-open if it occurs again.

closed 2019-10-07 by martiniil