PilzDE/pilz_robots #253: Add all packages to the package.xml in the metapackage

agutenkunst, 2019-09-26:


jschleicher, 2019-09-27:
You mean what is installed on `apt install ros-*-pilz-robots`?

* `pilz_testutils` is deliberately missing, since we do not want to pollute the end-user's installation with test-packages (only `test-depend` for other packages)
* prbt_gazebo is missing (as far as I remember), because it isn't required and moveit-fake-execution is enough for most users
* `pilz_control` and `prbt_hardware_support` are implicitly installed from `prbt_support`, see https://github.com/PilzDE/pilz_robots/blob/cb42db7b2f4510b1a7b2639e7e213fda824e0d96/prbt_support/package.xml#L30-L31 but it's probably good style to add them in the meta-package directly.

agutenkunst, 2019-10-08:
I came from a totally different perspective: http://wiki.ros.org/pilz_robots
And I wondered why the list on top is so short...

PR #267 (2019-10-10): Add packages to meta-package package.xml
Closes #253
closed 2019-10-10 by jschleicher