PilzDE/pilz_robots #299: Random chars at the end of firmware string

agutenkunst, 2020-01-14:
### Commit
db438915d632097fc1c29a9ebaefad27370f396b

### Steps to reproduce
Launch robot with real system

### Expected behavior
Firmware of each joint is displayed properly

### Observed behavior
Random chars are added to the firmware string

### Logfiles
```
[ INFO] [1578999408.520278926]: Firmware version [prbt_joint_1]: 100 Build:11158 Date:2018-06-07 16:49:55
[ INFO] [1578999408.520434190]: Firmware version [prbt_joint_2]: 100 Build:11158 Date:2018-06-07 16:49:55
[ INFO] [1578999408.520502982]: Firmware version [prbt_joint_3]: 100 Build:11158 Date:2018-06-07 16:49:55u
[ INFO] [1578999408.520559978]: Firmware version [prbt_joint_4]: 100 Build:11158 Date:2018-06-07 16:49:555
[ INFO] [1578999408.520618411]: Firmware version [prbt_joint_5]: 100 Build:11158 Date:2018-06-07 16:49:55E
[ INFO] [1578999408.520675839]: Firmware version [prbt_joint_6]: 100 Build:11158 Date:2018-06-07 16:49:55
```

agutenkunst, 2020-02-24:
@jschleicher could you provide more objects which are strings? I would like to see if they are all off-by-one, if yes this could be an issue with https://github.com/ros-industrial/ros_canopen if not this is on us.

jschleicher, 2020-02-25:
@agutenkunst The version registers 0x1008-0x100A are strings, see https://github.com/PilzDE/pilz_robots/blob/melodic-devel/prbt_support/config/prbt_0_1.dcf (grep for `DataType=0x0009`).

commit 0eecd0b3 (2020-01-14): Trim firmware string. Closes #299.
commit 7945e33a (2020-02-20): Trim firmware string. Closes #299.
commit d7d93f8c (2020-02-27): Trim firmware string. Closes #299. (#300)

* Trim firmware string. Workaround for #299
commit 7c7c37a8 (2020-03-12): Trim firmware string. Closes #299. (#300)

* Trim firmware string. Workaround for #299
closed 2020-02-27 by hslusarek