PilzDE/pilz_robots #301: CAN driver needs to be manually restarted after FS controller disconnect

HiddenBug, 2020-01-14:
### Observed behavior
If the connection to the FS controller (e.g. PSS 4000) is lost and, subsequently, established again, the CAN driver needs to be restarted via:
```
sudo ip link set can0 down
sudo ip link set can0 up type can bitrate 1000000
```
This statement remains true, even if the root-user changes the `/etc/network/interfaces` to automatically start the CAN driver via:
```
auto can0
iface can0 can static
        bitrate 1000000
```

### Problems
The described behavior leads to some unfavorable situations:
- If the CAN driver needs to be restarted, the "normal" robot user (which is not necessarily root) suddenly needs root rights. The question is: Is this the favored behavior?
- If the root user did change the  `/etc/network/interfaces` to automatically start the CAN driver: Why do we still need to manually restart the CAN driver? Why does is not happen automatically?

jschleicher, 2020-04-22:
Plugging off/on the USB cable (PCANusb) should re-initialize the driver. Maybe add a note to troubleshooting section of #339?

agutenkunst, 2020-04-30:
> Plugging off/on the USB cable (PCANusb) should re-initialize the driver. 

@tobi-v was able to confirm this on the real hardware. sudo ist not needed.

closed 2020-04-30 by agutenkunst