PilzDE/pilz_robots #304: Coverage for nose tests on travis

ct2034, 2020-01-31:
We can currently not run pure pyhton tests (i.e. nosetests) on travis.
I tried it here: https://github.com/PilzDE/pilz_robots/pull/285/commits/362ee70766419dcedc3f12e0b2b185b66e41f961
@agutenkunst  added instructions, how to run it locally here: https://github.com/PilzDE/pilz_robots/pull/285/commits/44fb88e0d26b9cd233d79d84767f02fbf8b9ebd6#diff-8a51f915a0ed0fa66528a45dfaeb5efcR36

agutenkunst, 2020-04-29:
Solved by https://github.com/mikeferguson/code_coverage ? @ct2034

jschleicher, 2020-04-29:
> Solved by https://github.com/mikeferguson/code_coverage ? @ct2034

No. That's just `rostest` which we currently use already. The problem of pure python tests remains.

jschleicher, 2020-06-23:
See PilzDE/pilz_industrial_motion#282 for how to integrate coverage for python nose tests.

closed 2020-06-23 by jschleicher