PilzDE/pilz_robots #309: Modbus Specification

JonathanGruner, 2020-02-12:
Could you please briefly describe the modbus specification here?
Not only the registers themselves, but also the contents of the registers, which value stands for which function. Also if this function is already available and how to activate it would not be bad. 
For example;

- Yaml-Parameter: OPERATION_MODE
- Register: 970
- 1-->Manual reduced speed, 2-->Manual High speed, 3-->Automatik
- prbt_moveitconfig: <arg name="has_operation_mode_support" default="false"/>
- not available yet/work in progress/released

ct2034, 2020-02-14:
Do you think it would make sense to add this information in e.g. [prbt_hardware_support/config/modbus_read_api_spec_pss4000.yaml](https://github.com/PilzDE/pilz_robots/blob/melodic-devel/prbt_hardware_support/config/modbus_read_api_spec_pss4000.yaml)? Yaml supports comments (https://www.tutorialspoint.com/yaml/yaml_comments.htm) so this should not be a problem.

JonathanGruner, 2020-02-14:
Definitely makes sense!
I just added it, see both pull requests.
I am not aware of the PNOZ Spec, maybe you can add the description in this file?

closed 2023-01-27 by martiniil