PilzDE/pilz_robots #310: Modbus read client

JonathanGruner, 2020-02-13:
Please replace the modbus_read **topic** with an **service**, where I can hand over the register number, and receive the content of this register.

ct2034, 2020-02-14:
Currently, this is implemented such that it repeatedly checks the defined registers and successively publishes the topic if the register values changed. I think this is a very neat solution. If we would change it to a service, all nodes requiring the data would need to repeatedly poll the service. I think is this not desirable in terms of traffic and usability. The publish-subscribe pattern (https://en.wikipedia.org/wiki/Publish%E2%80%93subscribe_pattern) works well in this scenario.

Maybe we can think of a service as an additional feature, though. Please reformulate the title accordingly. Also, this probably goes into the bigger idea of making the Modbus client more standalone.

JonathanGruner, 2020-02-14:
I like the idea of make the Modbus Client more standalone! We will talk about that in the next Meeting!
Currently I had some issues with two Modbus clients (the other one was pymodbus) --> They were somehow blocking each other.
I just recognized that I could copy the specification yaml files into my workspace and give the path as an argument to robot.launch. I guess with modifying those files, I am able to send and receive more registers? Is this foreseen from you like this?

JonathanGruner, 2020-02-14:
See Issue #250 
https://github.com/PilzDE/pilz_robots/issues/250

closed 2020-02-14 by JonathanGruner