PilzDE/pilz_robots #326: Travis fails for missing ros-melodic-moveit-ros-visualization

ct2034, 2020-04-03:
All build on the current _testing_ deb repo are failing. 
For example: https://travis-ci.org/github/PilzDE/pilz_robots/jobs/670125275

```
E: Unable to locate package ros-melodic-moveit-ros-visualization
ERROR: the following rosdeps failed to install
  apt: command [apt-get install -y -qq ros-melodic-moveit-ros-visualization] failed
```

ct2034, 2020-04-03:
The error happened on the buildfarm, too. But seems to be fixed there now. http://build.ros.org/job/Kdev__pilz_robots__ubuntu_xenial_amd64/

martiniil, 2020-04-06:
http://repositories.ros.org/status_page/ros_melodic_default.html?q=moveit_ros_visualization

https://github.com/ros-planning/moveit/issues/1996

martiniil, 2020-04-08:
Fixed: https://travis-ci.org/github/PilzDE/pilz_robots/builds/672431148

closed 2020-04-08 by martiniil