PilzDE/pilz_robots #327: Reorganisation of launch files

ct2034, 2020-04-03:
I think it is fairly confusing, especially for new users, that launch files in one package heavily influence stuff from other packages (and back). 
Especially `prbt_support/launch/robot.launch` which is the default way to configure the nodes from `prbt_hardware_support`. 
Same goes for `prbt_moveit_config/launch/moveit_planning_execution.launch`. 
(`prbt_moveit_config` has also way too many launch files. I doubt that even half of them are used by anyone.)
My suggestion would be to reorganize the packages according to the rule that packages should contain launch files that only configure (and for that matter _know about_) nodes in that package.

Which could lead for example to:
* `robot.launch` starts the robot, but only the robot (with can, no modbus). I support the idea of this being a file to _simply_ run the robot. But, that should be it. It should not be concerned with safety controllers and all the rest.
* some launch file in `prbt_hardware_support` starts the modbus connection and all the associated adapters (run permitted, brake test, operation mode ...)
* `prbt_moveit_config` ONLY contains the moveit configuration (with capabilities etc) but not also knows about the hardware
* We could either choose one of the above to be the central entry point or create a `prbt_bringup` package to start the full system, including all the 3 previously mentioned.

jschleicher, 2020-04-09:
I like the overall idea, but regarding your points 3 and 4 I'd like to object:

> * `prbt_moveit_config` ONLY contains the moveit configuration (with capabilities etc) but not also knows about the hardware

the `*_moveit_config` package is generated from and compatible with the moveit_setup_assistant. The files load e.g. the robot_description, joint and cartesian limits and semantic description, that cannot be hardware-independent :-(

Our `*_bringup` package IS `prbt_support`. During ROSIN we discussed with Nadia, if we should name our package `prbt_bringup` and her point was (as far as I remember), that bringup packages are sole launch file-packages. We have a driver with configuration for CANopen and thus are more like a 'driver+bringup' package that follows the naming `*_support`.

Still: Feel free to separate robot.launch into can and ethernet that can both be included either from a thin robot.launch or both individually from `moveit_planning_execution.launch`.

jschleicher, 2020-10-09:
Some of your points have been adressed and implemented in the meantime. Please (re-)open or better provide PR if you'd like to see further optimizations included.

closed 2020-10-09 by jschleicher