PilzDE/pilz_robots #329: Allow robot.launch without modbus

jschleicher, 2020-04-06:
### Commit
44a3d6b

### Steps to reproduce
1. create your own application package as in https://github.com/PilzDE/pilz_application_templates/
2. try to startup without modbus connection

### Expected behavior
bringup should be possible without modbus (at least for debugging whether a communication problem lies in can or modbus connection)

### Observed behavior
Startup always includes modbus.

From https://github.com/PilzDE/pilz_application_templates Readme:
It should not be necessary, to change the original launch files robot.launch or move_group.launch and thus fork pilz_robots. If there are any options missing, please file an Issue.

### Reference to test
Braketest and status_indicator can be disabled individually, but the modbus node is always present.

martiniil, 2020-04-06:
This sounds more like a missing feature. I think we can combine this with #327.

hslusarek, 2020-04-07:
@dbakovic is going to write a story for this issue. Therefore, I more or less consider this issue closed.

martiniil, 2020-04-21:
@jschleicher 
- Should the user be able to command robot movements in general?
- Should the user be able to command the robot via the Python API?

jschleicher, 2020-04-29:
Solved with #353

commit 3000814c (2020-04-09): allow skipping modbus altogether

Fixes #329
closed 2020-04-29 by jschleicher