PilzDE/pilz_robots #337: ROSException: timeout exceeded while waiting for service /prbt/get_speed_override

tswie, 2020-04-09:
### Commit
Latest melodic push.

### Steps to reproduce
This happens when I attempt to run a motion server for my robot using PILZ. 

### Expected behavior
The server should launch and await requests from other programs.

### Observed behavior
The initialization times out waiting for the /prbt/get_speed_override service. This code previously worked on kinetic. 

Which node is supposed to run this service exactly? I can't find it anywhere.

hslusarek, 2020-04-15:
The `/prbt/get_speed_override service` also exists in `kinetic-devel` see [operation_mode_setup_executor_node.cpp](https://github.com/PilzDE/pilz_robots/blob/73d2dc0aa4ce163025bfa24710f51d6063f55338/prbt_hardware_support/src/operation_mode_setup_executor_node.cpp#L80) on `kinetic-devel`.

jschleicher, 2020-04-22:
@tswie The service is provided by the modbus communcation nodes when running a real PRBT hardware setup including safety controller. Without operationmode feature just launch `fake_speed_override_node` as in https://github.com/PilzDE/pilz_robots/blob/melodic-devel/prbt_hardware_support/launch/fake_safety_interface.launch.

closed 2020-04-22 by jschleicher