PilzDE/pilz_robots #339: Write Tutorial

ct2034, 2020-04-21:


hslusarek, 2020-04-22:
Created empty tutorial page: [pilz_robots/Tutorials/HowToSetupTheRobot](https://wiki.ros.org/pilz_robots/Tutorials/HowToSetupTheRobot)

martiniil, 2020-04-24:
Files for the tutorial: https://github.com/PilzDE/pilz_tutorials/pull/12/files

SansoneG, 2020-04-28:
Consider using [GetTaggedCode](http://library.isr.ist.utl.pt/docs/roswiki/WikiMacros.html#GetCode.2C_GetTaggedCode) to keep the tutorials synced with github repo.
Also https://github.com/PilzDE/pilz_tutorials/pull/12#issue-408381844

SansoneG, 2020-04-28:
[Model your first application using the Pilz Manipulator Module PRBT](https://wiki.ros.org/pilz_robots/Tutorials/ModelYourApplicationWithPRBT) -> This is listed as `Next Tutorial` and under `Prerequisites`

agutenkunst, 2020-04-28:
> [Model your first application using the Pilz Manipulator Module PRBT](https://wiki.ros.org/pilz_robots/Tutorials/ModelYourApplicationWithPRBT) -> This is listed as `Next Tutorial` and under `Prerequisites`

If fixed this.

closed 2020-04-30 by SansoneG