PilzDE/pilz_robots #343: Review

ct2034, 2020-04-21:


jschleicher, 2020-04-28:
Some comments from my side: 
[tutorial_real_robot_01_.pdf](https://github.com/PilzDE/pilz_robots/files/4544206/tutorial_real_robot_01_.pdf)

PilzES, 2020-04-28:
I've got a few suggestions and questions: 

- I guess from the provided example that It is now possible to run the robot *without* the safety PLC. There should be a warning somewhere indicating clearly that this is _not_ a production-ready method. I know there's one at the beginning about system integrators, but maybe adding a comment when flagging `<arg name="iso10218_support" value="false"/>` would help clarify. 

- I strongly suggest adding a safety section (connecting to the pss4000, making sure you are recieving the modbus signals, and so on). 

- This is supposed to be a follow-up to the previous tutorials. In that vein, It would be convenient to show how to run the real robot **while taking into account the project that has been built before**. This means loading the URDF files and the environment that was previously described (which is especially important when running on a real system for collision detection).

SansoneG, 2020-04-28:
My comments https://github.com/PilzDE/pilz_robots/issues/339#issuecomment-620424958 and https://github.com/PilzDE/pilz_robots/issues/339#issuecomment-620429151

jschleicher, 2020-04-28:
@PilzES Thanks for your comment! More tutorials regarding a complete 10218 safety compliant application are in the Backlog.
So we deliberately defer your second point to the next tutorial(s), focusing on just the CAN connection here.

agutenkunst, 2020-04-28:
@jschleicher  thanks for your input

> Some comments from my side:
> [tutorial_real_robot_01_.pdf](https://github.com/PilzDE/pilz_robots/files/4544206/tutorial_real_robot_01_.pdf)

I addressed the following from your comments so far:
- Remove "Next tutorial" which was simply wrong
- Merge Warning Boxes
- Replace "CERTAIN" with "SURE"
- Remove "Basic ROS knowledge." since this is redundant with the requirement for the other tutorial
- Remove the manual can bringup section
- Add a description of the desired effect of the "roslaunch" (rviz starts)
- Add some more required hardware to the perquisites

I did not address:
- Replace "steps 1-4 of" with "steps 1 and 2" since later it is referred to a *pilz_tutorials* packages which is created at step 4

Additionally I:
- Added a reboot step and check for *can0* inside the CAN Section

See http://wiki.ros.org/action/diff/pilz_robots/Tutorials/HowToSetupTheRobot?action=diff&rev1=17&rev2=18 for the changes.

agutenkunst, 2020-04-28:
> I've got a few suggestions and questions:
>
> * I guess from the provided example that It is now possible to run the robot _without_ the safety PLC. There should be a warning somewhere indicating [...]

Hi @PilzES 
is http://wiki.ros.org/action/diff/pilz_robots/Tutorials/HowToSetupTheRobot?action=diff&rev1=18&rev2=19 ok for you?

agutenkunst, 2020-04-28:
> @PilzES Thanks for your comment! More tutorials regarding a complete 10218 safety compliant application are in the Backlog.
> So we deliberately defer your second point to the next tutorial(s), focusing on just the CAN connection here.

@jschleicher does this also address this comment from @PilzES ?

> * This is supposed to be a follow-up to the previous tutorials. In that vein, It would be convenient to show how to run the real robot **while taking into account the project that has been built before**. This means loading the URDF files and the environment that was previously described (which is especially important when running on a real system for collision detection). Perhaps adding a `sim` argument to the previous launch file or creating one with a more clear name `launch_real_system`.

closed 2020-04-30 by agutenkunst