PilzDE/pilz_robots #356: rosdep cannot find definition for `pilz_status_indicator_rqt`

SansoneG, 2020-04-23:
### Commit
9aec9cd6b

### Steps to reproduce
1. run `rosdep check prbt_support`

### Expected behavior
List all missing system dependencies for package `prbt_support`

### Observed behavior
Missing system dependencies are displayed fine except for `pilz_status_indicator_rqt`

### Logfiles
`rosdep check prbt_support` yields:
```
System dependencies have not been satisfied:
apt	ros-melodic-pilz-testutils
apt	ros-melodic-pilz-utils
apt	ros-melodic-pilz-control
apt	ros-melodic-prbt-hardware-support
ERROR[prbt_support]: Cannot locate rosdep definition for [pilz_status_indicator_rqt]
	rosdep key : pilz_status_indicator_rqt
	OS name    : ubuntu
	OS version : bionic
	Data:
<no data>
```

`apt-cache search pilz rqt` yields:
```
ros-melodic-pilz-status-indicator-rqt - Showing information about operation mode, status and speed override of the robot.
```

### Additional Info
`apt update` and `apt upgrade` didn't solve the issue. `ros-melodic-pilz-status-indicator-rqt` is in the `apt-cache`

agutenkunst, 2020-05-04:
Tested with b5d88cf6b3d0628ec25a5720522ed4a7d213c4fe

Cannot reproduce.

`rosdep check prbt_support` yields:

```
System dependencies have not been satisfied:
apt	ros-melodic-pilz-utils
apt	ros-melodic-pilz-control
apt	ros-melodic-pilz-status-indicator-rqt
apt	ros-melodic-prbt-hardware-support
```

agutenkunst, 2020-05-04:
Also cannot reproduce with 9aec9cd

agutenkunst, 2020-05-04:
@SansoneG Close ok?

SansoneG, 2020-05-04:
I investigated again. Running `rosdep update` solved the problem. Thanks

closed 2020-05-04 by SansoneG