PilzDE/pilz_robots #379: Modbus API Break?

martiniil, 2020-05-06:
### Commit
b5d88cf6b3d0628ec25a5720522ed4a7d213c4fe

### Steps to reproduce
```
roslaunch prbt_moveit_config moveit_planning_execution.launch sim:=false
rosservice call /prbt/execute_braketest
rosservice call /prbt/brake_test_required
```

### Expected behavior
Result is `-1`

### Observed behavior
Result is `1`, like before the braketest.

martiniil, 2020-05-07:
In #312 the new api should be documented.

ct2034, 2020-05-07:
The values that correlate to the operation modes have also changed. In our control cabinet (in relation to the labels on pitmode)

martiniil, 2020-05-08:
I started implementing the changes: #385

jschleicher, 2020-05-12:
Is this break on ROS side or on PSS4k side? `-1` looks more like a bug in the implementation (uninitialized) than an update to the specification needing a new version?

jschleicher, 2020-06-17:
Implemented in #385

closed 2020-06-17 by jschleicher