PilzDE/pilz_robots #382: Modbus connection without active network produces no error message

ct2034, 2020-05-07:
### Commit
b5d88cf6b3d0628ec25a5720522ed4a7d213c4fe

### Steps to reproduce
1. make sure modbus connection can be established.
2. disconnect ethernet cable
3. try to establish modbus connection again

### Expected behavior
An error message is shown that describes the problem directly

### Observed behavior
No error is produced. In fact, the console looks exactly the same as when a connection is established successfully.

agutenkunst, 2020-06-10:
I check this and observed that if you run with some arbitrary IP 
```
roslaunch prbt_hardware_support modbus_client.launch modbus_server_ip:=210.168.0.1 modbus_server_port:=5020
```
it will after some time throw an error (for me 130s)

This is due getting stuck at https://github.com/PilzDE/pilz_robots/blob/cddde81f7ec3f4ab8c25703cb04c034aa75af869/prbt_hardware_support/src/libmodbus_client.cpp#L40

In my opinion this could be fixed by simple calling `setResponseTimeout` before connecting, however since Ubuntu 18 comes with libmodbus 3.0.6 this has no effect. We would need https://github.com/stephane/libmodbus/commit/8dc21ddaf51bace2030f5f06b51754eb4de6e87d

But maybe after https://github.com/PilzDE/pilz_robots/blob/cddde81f7ec3f4ab8c25703cb04c034aa75af869/prbt_hardware_support/src/libmodbus_client.cpp#L38 we could use this `modbus_connection_` to manually set the timeout (maybe SO_RCVTIMEO and SO_SNDTIMEO via setsockopt) for the raw socket. But I am not sure if this influences the actual timeout during the initial commit. (See https://stackoverflow.com/questions/4181784/how-to-set-socket-timeout-in-c-when-making-multiple-connections)

This observation can be reproduced without a actual server.

If you need a server for further investigation maybe https://gist.github.com/agutenkunst/fc2d260ef9aabe34d8a9e8b3e19ae857 helps.

jschleicher, 2020-06-17:
Currently in progress, see #410

commit a2ea32b4 (2020-06-19): Improving error response in case of IP adress failures. (#410)

Add connetion precheck for modbus connection. Closes #382 

* connection precheck with timeout before establishing modbus connection for better error output.
closed 2020-06-19 by agutenkunst