PilzDE/pilz_robots #400: Review acceptance tests

martiniil, 2020-05-26:


agutenkunst, 2020-05-27:
I checked 
https://github.com/PilzDE/pilz_robots/blob/49d3e03b191f84367f583cf9f9e0505e020b5cef/pilz_control/test/acceptance_test_speed_monitoring.md

https://github.com/PilzDE/pilz_robots/blob/49d3e03b191f84367f583cf9f9e0505e020b5cef/pilz_control/test/acceptance_test_speed_monitoring.py
and had some remarks.

See https://github.com/PilzDE/pilz_robots/pull/308#pullrequestreview-419066442

agutenkunst, 2020-05-27:
Reassign on @martiniil

martiniil, 2020-05-27:
Why did you skip the other files? `joint_states_speed_observer.*`?

agutenkunst, 2020-05-27:
> @agutenkunst Why did you skip the other files? `joint_states_speed_observer.*`?

Sry missed. Now added comments.

See especially https://github.com/PilzDE/pilz_robots/commit/49d3e03b191f84367f583cf9f9e0505e020b5cef#r39482161

closed 2020-06-19 by hslusarek