PilzDE/pilz_robots #415: fatal error: 'gmock/gmock.h' file not found

v4hn, 2020-06-19:
### Commit

`melodic-devel`

### Steps to reproduce

1. Probably only by setting up a non-debian Linux
2. Build

### Observed behavior

```
pilz_robots/pilz_testutils/src/joint_state_publisher_mock.cpp:23:10: fatal error: 'gmock/gmock.h' file not found
```

Looking at the respective `CMakeLists.txt`, `GMOCK_INCLUDE_DIRS` is not mentioned anywhere.
In another project I had to switch from `catkin_add_gtest` to `catkin_add_gmock` to fix this, but as this is a library, that wouldn't work. You probably have to add gmock manually.

I'm guessing Ubuntu/Debian might provide a symlink `/usr/include/gmock`?

hslusarek, 2020-06-29:
Thanks for your input. I looked at the corresponding test and discovered that the include is actually not needed. I opened a PR #425 to solve this issue. If you like you can try it out and see if it works for you. Thx and Peace.

hslusarek, 2020-06-29:
Ok, I also opened PR #426 which should add the missing includes, etc.

@v4hn Maybe you can try it out and see if it works for you.

v4hn, 2020-06-29:
Sorry, but I'm confused. Why do you propose to add the dependencies when the include is not necessary?

hslusarek, 2020-06-30:
> Sorry, but I'm confused. Why do you propose to add the dependencies when the include is not necessary?

Sorry, for the confusing communication. The `joint_state_publisher_mock.cpp` file does _not_ need the `gmock` include. However, the `async_test` does need the `gmock` include, although it's missing the corresponding `gmock` include right now. I'm going to open a PR for this, too.

**Note:**
The `async_test` does need the `gmock` include because it uses `::testing::InvokeWithoutArgs` (which is part of the gmock framework) in [line 107](https://github.com/PilzDE/pilz_robots/blob/672a3f2bde4a77cc3953737a3e809aa6a00bd512/pilz_testutils/include/pilz_testutils/async_test.h#L107)   . That  [`::testing::InvokeWithoutArgs`] is part of the gmock framework can be seen on [github.com/googletest](https://github.com/google/googletest/blob/master/googlemock/include/gmock/gmock-actions.h#L1261).

v4hn, 2020-06-30:
> The `async_test` does need the `gmock` include because it uses `::testing::InvokeWithoutArgs`

Syntactically it does not need the include because it only defines preprocessor macros with the offending symbol, it does not use them.
As far as I can see, the whole library does not use a single gtest or gmock symbol in preprocessed code.

So of course you can add the dependencies, but everything is fine without them as well.
The only thing you gain would be a gmock include in `async_test.h`, which the user usually provides on their own anyway.
Because this is only about preprocessor directives, it wouldn't even require a specific include order between them.

Joint pointing it out though, as long as it's resolved I don't mind either solution.

hslusarek, 2020-07-01:
PR #430 is going to close this issue. Thx to everyone for the participation in the discussions.

closed 2020-07-07 by hslusarek