PilzDE/pilz_robots #418: Planner plans trajectories that exceed acceleration limit.

SansoneG, 2020-06-22:
### Commit
cb782c272ee6

### Steps to reproduce
How can the issue/bug be reproduced?
1. Switch into AUTO mode
2. Use RViz to plan a movement with velocity- and acceleration scaling 1.0.
3. Execute the movement.

### Expected behavior
Planning and Executing works without error.

### Observed behavior
Planning works, but executing fails. The robot doesn't even start moving.

### Logfiles
```
[ INFO] [1592834842.386682498]: Recovering successful
[ INFO] [1592834848.094568331]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1592834848.094682587]: Initialized Point-to-Point Trajectory Generator.
[ INFO] [1592834848.094722985]: Generating PTP trajectory...
[ INFO] [1592834849.925210143]: Execution request received
[ERROR] [1592834849.972380926]: Acceleration limit violated by joint prbt_joint_6. Desired acceleration: 4.43134rad/s^2, limit:  4rad/s^2.
[ INFO] [1592834849.972646862]: Controller prbt/manipulator_joint_trajectory_controller/ successfully finished
[ WARN] [1592834849.972706115]: Controller handle prbt/manipulator_joint_trajectory_controller/ reports status PREEMPTED
[ INFO] [1592834849.972743617]: Completed trajectory execution with status PREEMPTED ...
[ INFO] [1592834849.972813891]: Execution completed: PREEMPTED
[ INFO] [1592834850.003672288]: PREEMPTED: Preempted
```
### Reference to test
https://github.com/PilzDE/pilz_robots/blob/melodic-devel/pilz_control/test/acceptance_test_acceleration_limit.md

martiniil, 2020-06-30:
@jschleicher @hslusarek I would like to discuss if we could skip the stop trajectory completely and let the firmware do the stopping. Then we would not need the acceleration limit, either.

jschleicher, 2020-06-30:
I'm not sure if, the firmware offers a stopping command in the meantime. My latest information is, that "halt" does close the brakes and thus wears out the mechanics, which has only limited numbers of "STOP-0" cycles.

martiniil, 2020-06-30:
But if we abort the trajectory and stay on the latest position, the firmware will stop the movement asap without closing the brakes?

hslusarek, 2020-06-30:
> But if we abort the trajectory and stay on the latest position, the firmware will stop the movement asap without closing the brakes?

I'm not sure if I understand your suggestion correctly, because your suggestion will cause a massive jerk if the robot moved with full speed beforehand. For me a strong/massive jerk does not sound good. But maybe I misunderstand your suggestion?

martiniil, 2020-06-30:
> > But if we abort the trajectory and stay on the latest position, the firmware will stop the movement asap without closing the brakes?
> 
> I'm not sure if I understand your suggestion correctly, because your suggestion will cause a massive jerk if the robot moved with full speed beforehand. For me, a strong/massive jerk does not sound good, especially with regard to the mechanical components of the robot. But maybe I misunderstand your suggestion?

Yes, you understood correctly. But:
- As far as I remember, the firmware is responsible for avoiding damage to the mechanics.
- In our case, the robot would not be allowed to move with full speed, only reduced.

jschleicher, 2020-06-30:
The joint values sent by the controller should obey the joint limits for both velocity and acceleration. That's what the configuration values are meant for.

But this issue is about planned trajectories that violate the acceleration limit in the controller - is there any connection to the stop trajectory, that I'm missing? Or is your suggestion independent from our current issue?

martiniil, 2020-06-30:
> The joint values sent by the controller should obey the joint limits for both velocity and acceleration. That's what the configuration values are meant for.
>
> But this issue is about planned [...]

My suggestion is connected to the current issue, because I questioned the acceleration limit which was only introduced due to a poor stop trajectory. I know, I question a lot...
But it is answered now. Thank you.

hslusarek, 2020-07-01:
> I know, I question a lot...

I think this is a good quality for a developer. :+1:

martiniil, 2020-07-10:
This is the desired velocity the controller produces from a PTP trajectory. You can see that the acceleration varies at the beginning:

[image]

martiniil, 2020-07-10:
I think the trajectories which are generated by the planner do not violate the limit. But through its interpolation routine the controller can slightly deviate from the intended trajectory.

I see possible solutions:
- Increase the limit for the controller/Decrease limit for planners (how much?)
- Tweak the planners. Some aspects:
  - Is it really necessary/helpful to specify velocities/accelerations instead of just positions? What is the effect of each option?
  - Velocity profile: Can it be tweaked?
- Fix stop trajectory such that the acceleration limit is no longer needed.

jschleicher, 2020-07-13:
The first option sounds like a quick fix. So I'd prefer to increase the limit inside the controller by 5-10% for now. I'll add a PBI to determine the actual limits to achieve 250mm/s incl. the delay.

ct2034, 2020-07-13:
@SansoneG can you please elaborate on the original issue: You are in Auto mode, right? Then we should not care about the limits.

SansoneG, 2020-07-13:
> @SansoneG can you please elaborate on the original issue: You are in Auto mode, right? Then we should not care about the limits.

Yes, the behaviour was observed in AUTO mode. The acceleration and velocity limits imposed by the robot hardware should not be exceeded even in AUTO mode. I think this is what was meant [here](https://github.com/PilzDE/pilz_robots/issues/418#issuecomment-651759617).
But by planning a movement with seemingly accepted acceleration values, the controller stops the movement. That is the issue.

ct2034, 2020-07-13:
Yes, the planner planning a trajectory that not violates the limits is one thing. We use the speed override for this: https://github.com/PilzDE/pilz_robots/blob/81b2dff62e9142139850a58e718d13d5355cfb97/prbt_hardware_support/src/operation_mode_setup_executor.cpp#L43

But I think we do not evaluate the operation mode when checking the acceleration limit: https://github.com/PilzDE/pilz_robots/blob/81b2dff62e9142139850a58e718d13d5355cfb97/pilz_control/include/pilz_control/pilz_joint_trajectory_controller_impl.h#L326
We check it when evaluating the speed limit: https://github.com/PilzDE/pilz_robots/blob/81b2dff62e9142139850a58e718d13d5355cfb97/pilz_control/src/cartesian_speed_monitor.cpp#L91

SansoneG, 2020-07-13:
I understand where you are coming from now.
If I understand correctly we cannot simply ignore the acceleration limit in AUTO mode, because the acceleration limit also serves the purpose of not allowing crazy stop trajectories.

martiniil, 2020-07-14:
> Yes, the planner planning a trajectory that not violates the limits is one thing. We use the speed override for this:
> 
> https://github.com/PilzDE/pilz_robots/blob/81b2dff62e9142139850a58e718d13d5355cfb97/prbt_hardware_support/src/operation_mode_setup_executor.cpp#L43

The purpose of setting the speed override to `0.1` is limiting the cartesian speed in T1. The acceleration limit is another matter. It should never be violated such that the hardware is not damaged.

martiniil, 2020-07-14:
> The first option sounds like a quick fix. So I'd prefer to increase the limit inside the controller by 5-10% for now. I'll add a PBI to determine the actual limits to achieve 250mm/s incl. the delay.

The quick fix should work for now. Imho the time and effort we would spend in determining "good" limits (I think it is not straightforward) would be better invested in a proper solution.

ct2034, 2020-07-14:
> The acceleration limit is another matter. It should never be violated such that the hardware is not damaged.

Okay, I got it. Thanks

martiniil, 2020-07-16:
For better understanding this picture shows the desired joint velocity of a single joint at the beginning of a PTP trajectory (Acceleration limits disabled).

The acceleration with which the trajectory is planned is exceeded in the first segment.

[image]

martiniil, 2020-07-16:
@jschleicher For the demo program we need more than 100% increase of the limit inside the controller. See https://github.com/PilzDE/pilz_robots/pull/442