PilzDE/pilz_robots #427: API Version Error

CloudYin, 2020-06-30:
### Commit
Released version (after sudo apt upgrade)

### Steps to reproduce
How can the issue/bug be reproduced?
1. sudo apt update & sudo apt upgrade
2. In PNOZmulti m B0, set the register 513 to "2"
3. In the robot launch file, set:
    <arg name="safety_hw" value="pnoz" />
    <arg name="iso10218_support" value="true" />
    <arg name="has_braketest_support" value="false" />
    <arg name="has_operation_mode_support" value="false" />
    <arg name="visual_status_indicator" value="false" />
3. Launch the robot.

### Expected behavior
Robot should be launched without error.

### Observed behavior
Robot cannot be moved because of the API version error. Settings in PNOZmulti has been double checked by Techsupport. 

### Logfiles
log file: /home/dell/.ros/log/c386bad4-baa2-11ea-b011-6cb311523476/robot_init-8*.log
[ERROR] [1593501884.409866452]: Received Modbus message of unsupported API Version: 0, required Version: 2

### Reference to test
If manually use other pilz robot package in the application package, e.g. pilz_robots-JonathanGruner-comments, such error will not occur.

hslusarek, 2020-07-07:
@jschleicher Ping

CloudYin, 2020-07-21:
Closed via #444

closed 2020-07-21 by CloudYin