PilzDE/pilz_robots #433: acceptance_test_speed_monitoring fails when executed on real robot

hslusarek, 2020-07-02:
### Commit
9aa67f91c75c1fd240d0562a6ef31a49bb264db9 on melodic-devel

### Steps to reproduce
Test fails in T1 at step 2 of `acceptance_test_speed_monitoring.md`

### Observed behavior
Test fails in T1 at step 2 of `acceptance_test_speed_monitoring.md` with:
```rosrun pilz_control acceptance_test_speed_monitoring.py
[ROSUNIT] Outputting test results to /home/hslusarek/.ros/test_results/pilz_control/rosunit-acceptance_test_speed_monitoring.xml
[INFO] [1593613437.052367]: Move to start position
[INFO] [1593613447.179552]: Test speed monitoring in T1 mode
[INFO] [1593613447.181057]: Determine target velocity for reaching the speed limit
[INFO] [1593613454.805999]: Move to start position
[Testcase: test_reduced_speed_mode] ... FAILURE!
FAILURE: Speed limit of 0.25[m/s] was violated
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/home/hslusarek/catkin_ws/src/pilz_robots/pilz_control/test/acceptance_test_speed_monitoring.py", line 149, in test_reduced_speed_mode
    self.assertGreater(_SPEED_LIMIT, self._max_frame_speed.get(), 'Speed limit of 0.25[m/s] was violated')
  File "/usr/lib/python2.7/unittest/case.py", line 943, in assertGreater
    self.fail(self._formatMessage(msg, standardMsg))
  File "/usr/lib/python2.7/unittest/case.py", line 410, in fail
    raise self.failureException(msg)
--------------------------------------------------------------------------------
-------------------------------------------------------------
SUMMARY:
 * RESULT: FAIL
 * TESTS: 1
 * ERRORS: 0 []
 * FAILURES: 1 [test_reduced_speed_mode]

```

### Expected behavior
Test should run at real robot without problems.

### Logfiles
```
roslaunch prbt_moveit_config moveit_planning_execution.launch sim:=False pipeline:=pilz_command_planner
... logging to /home/hslusarek/.ros/log/5f3a74ee-bba6-11ea-9122-84a93e5e04ca/roslaunch-titan-20147.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
xacro: in-order processing became default in ROS Melodic. You can drop the option.
xacro: in-order processing became default in ROS Melodic. You can drop the option.
xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://titan:37631/

SUMMARY
========

CLEAR PARAMETERS
 * /prbt/driver/

PARAMETERS
 * /controller_joint_names: ['prbt_joint_1', ...
 * /move_group/allow_trajectory_execution: True
 * /move_group/capabilities: pilz_trajectory_g...
 * /move_group/controller_list: [{'action_ns': 'f...
 * /move_group/disable_capabilities: 
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planning_plugin: pilz::CommandPlanner
[... 338 lines cut ...]
[ERROR] [1593613393.286840231]: EMCY received: 86#0000000000000000
[ERROR] [1593613393.295313747]: EMCY received: 87#0000000000000000
[ERROR] [1593613393.299689084]: EMCY received: 88#0000000000000000
[ INFO] [1593613393.300049691]: Recovering successful
[ INFO] [1593613393.681632375]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1593613393.682189400]: Planning attempt 1 of at most 1
[ INFO] [1593613393.682524884]: Initialized Point-to-Point Trajectory Generator.
[ INFO] [1593613393.682771572]: Generating PTP trajectory...
[ INFO] [1593613393.688062512]: Disabling trajectory recording
[ INFO] [1593613395.208894865]: Controller prbt/manipulator_joint_trajectory_controller/ successfully finished
[ INFO] [1593613395.250876023]: Completed trajectory execution with status SUCCEEDED ...

[ INFO] [1593613401.621576292]: Halting down...
[ INFO] [1593613401.621636047]: Halting down successful
[ INFO] [1593613407.577709618]: Recovering...
[ERROR] [1593613407.643614772]: EMCY received: 83#0000000000000000
[ERROR] [1593613407.647785947]: EMCY received: 84#0000000000000000
[ERROR] [1593613407.651528545]: EMCY received: 85#0000000000000000
[ERROR] [1593613407.655049547]: EMCY received: 86#0000000000000000
[ERROR] [1593613407.659547048]: EMCY received: 87#0000000000000000
[ERROR] [1593613407.663222472]: EMCY received: 88#0000000000000000
[ INFO] [1593613407.663290191]: Recovering successful
[ INFO] [1593613489.501810698]: Halting down...
[ INFO] [1593613489.501871690]: Halting down successful
```

martiniil, 2020-07-02:
I had a quick look and found something which might cause the failure: #436

martiniil, 2020-07-29:
Solved: 7abce15b1dd9676b8abc737aa67265f1ac2ff23c

closed 2020-07-29 by martiniil