PilzDE/pilz_robots #434: acceptance_test_acceleration_limit fails when executed on real robot

hslusarek, 2020-07-02:
### Commit
9aa67f91c75c1fd240d0562a6ef31a49bb264db9 on melodic-devel

### Observed behavior
Test fails in step 3 of the test.

### Steps to reproduce
- Execute acceptance test as described in `acceptance_test_acceleration_limit.md` until step 3 (including step 3).

### Observed behavior
- Everything went fine (as described in the test) until Rviz motion with velocity- and acceleration scaling set to 1.0 had to be executed. Then ROS refused to execute motion because the controller is in hold-mode. 
  - Manually releasing the hold-mode and reducing the (velocity- and acceleration scaling because otherwise trajectory with too high limits are planned) allows the robot to be moved again. 
  - For more information see the terminal log which is printed below.

### Expected behavior
- How should the user relase the hold-mode in operation mode AUTOMATIC?
    - I suggest to release the hold-mode via the "Reset" button because from a user perspective this makes sense. But this is currently not possible.

### Logfiles
```
roslaunch prbt_moveit_config moveit_planning_execution.launch sim:=false pipeline:=pilz_command_planner
... logging to /home/hslusarek/.ros/log/9152428e-bba3-11ea-9122-84a93e5e04ca/roslaunch-titan-19535.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
xacro: in-order processing became default in ROS Melodic. You can drop the option.
xacro: in-order processing became default in ROS Melodic. You can drop the option.
xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://titan:35169/

SUMMARY
========

CLEAR PARAMETERS
 * /prbt/driver/

PARAMETERS
 * /controller_joint_names: ['prbt_joint_1', ...
 * /move_group/allow_trajectory_execution: True
 * /move_group/capabilities: pilz_trajectory_g...
 * /move_group/controller_list: [{'action_ns': 'f...
 * /move_group/disable_capabilities: 
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planning_plugin: pilz::CommandPlanner
 * /move_group/planning_scene_monitor/publish_geometry_updates: True
 * /move_group/planning_scene_monitor/publish_planning_scene: True
 * /move_group/planning_scene_monitor/publish_state_updates: True
 * /move_group/planning_scene_monitor/publish_transforms_updates: True
 * /move_group/request_adapters: 
 * /move_group/start_state_max_bounds_error: 0.1
 * /move_group/trajectory_execution/allowed_execution_duration_scaling: 1.2
 * /move_group/trajectory_execution/allowed_goal_duration_margin: 0.5
 * /move_group/trajectory_execution/allowed_start_tolerance: 0.01
 * /prbt/driver/bus/device: can0
 * /prbt/driver/defaults/eds_file: config/prbt_0_1.dcf
 * /prbt/driver/defaults/eds_pkg: prbt_support
 * /prbt/driver/name: manipulator
 * /prbt/driver/nodes/prbt_joint_1/braketest_required: True
 * /prbt/driver/nodes/prbt_joint_1/id: 3
 * /prbt/driver/nodes/prbt_joint_2/braketest_required: True
 * /prbt/driver/nodes/prbt_joint_2/id: 4
 * /prbt/driver/nodes/prbt_joint_3/braketest_required: True
 * /prbt/driver/nodes/prbt_joint_3/id: 5
[... 354 lines cut ...]
[ERROR] [1593612241.523679203]: EMCY received: 83#0000000000000000
[ERROR] [1593612241.526909131]: EMCY received: 84#0000000000000000
[ERROR] [1593612241.529494222]: EMCY received: 85#0000000000000000
[ERROR] [1593612241.533150226]: EMCY received: 86#0000000000000000
[ERROR] [1593612241.536662769]: EMCY received: 87#0000000000000000
[ERROR] [1593612241.538683653]: EMCY received: 88#0000000000000000
[ INFO] [1593612241.538724697]: Recovering successful
[ERROR] [1593612259.550584597]: Acceleration limit violated by joint prbt_joint_2. Desired acceleration: 8024.73rad/s^2, limit:  4rad/s^2.
[ERROR] [1593612272.640684136]: Acceleration limit violated by joint prbt_joint_2. Desired acceleration: 7732.69rad/s^2, limit:  4rad/s^2.
[ INFO] [1593612296.971693035]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1593612296.971784275]: Initialized Point-to-Point Trajectory Generator.
[ INFO] [1593612296.971823190]: Generating PTP trajectory...
[ INFO] [1593612302.901140614]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1593612302.901236473]: Initialized Point-to-Point Trajectory Generator.
[ INFO] [1593612302.901302282]: Generating PTP trajectory...
[ INFO] [1593612304.981566220]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1593612304.981651626]: Planning attempt 1 of at most 1
[ INFO] [1593612304.981728608]: Initialized Point-to-Point Trajectory Generator.
[ INFO] [1593612304.981780534]: Generating PTP trajectory...
[ WARN] [1593612305.011915726]: Controller received new commands but won't react because it is currently in holding mode.
[ WARN] [1593612305.012149338]: Controller prbt/manipulator_joint_trajectory_controller/ failed with error INVALID_GOAL: 
[ WARN] [1593612305.012258652]: Controller handle prbt/manipulator_joint_trajectory_controller/ reports status FAILED
[ INFO] [1593612305.012311569]: Completed trajectory execution with status FAILED ...
[ INFO] [1593612305.035716991]: ABORTED: Solution found but controller failed during execution
```

martiniil, 2020-07-02:
```
I suggest to release the hold-mode via the "Reset" button because from a user perspective this makes sense. But this is currently not possible.
```
Yes, I had this problem, too. You can use the Emergency-button+Reset as described here: https://github.com/PilzDE/robot_lab/blob/master/robot_instructions.md

martiniil, 2020-07-02:
So, did the test fail due to #418?

hslusarek, 2020-07-02:
> So, did the test fail due to #418?

No, the test already failed beforehand. The controller was in hold-mode.

hslusarek, 2020-07-02:
> You can use the Emergency-button+Reset as described

My question is: Is that the general advice for a user which encounters the problem in a real application in AUTOMATIC mode?

jschleicher, 2020-07-03:
The current "experimental" PSS4k program requires that you press ESTOP once, before you can "Reset". Will be fixed for the final safety program.
Closing here.

closed 2020-07-03 by jschleicher