PilzDE/pilz_robots #435: Status indicator GUI shows red icons

hslusarek, 2020-07-02:
### Commit
9aa67f91c75c1fd240d0562a6ef31a49bb264db9 on melodic-devel

### Steps to reproduce
1. Start the ROS system as described in the [pilz_robots/README.md](https://github.com/PilzDE/pilz_robots/blob/melodic-devel/README.md#running-on-the-real-robot)
2. After start-up the status indicator GUI shows constantly red icons for ROS and the hardware:

[image]

### Observed behavior
- As a user I get easily confused by a GUI which constantly shows me that something is not okay.

### Expected behavior
- Maybe it would be better, if the feature is not supported, to deactivate this icons or at least to make them grey or something?

jschleicher, 2020-07-03:
Thanks for reporting! That's a regression, we should investigate.

martiniil, 2020-07-29:
Solved: f87451ea5f18eed16dc3287c7d946a25bf2be3cf

closed 2020-07-29 by martiniil