PilzDE/pilz_robots #439: Cyclomatic dependency between prbt_moveit_config and prbt_ikfast_manipulator_plugin

agutenkunst, 2020-07-09:
### Commit
4be09959c4c7c85872a34b8f0ee66516c1bb5a81

### Reason
We have a dependency I guess because we need the plugin If we launch
https://github.com/PilzDE/pilz_robots/blob/4be09959c4c7c85872a34b8f0ee66516c1bb5a81/prbt_moveit_config/package.xml#L33

and we would need a test-dependency the other way around due to
https://github.com/PilzDE/pilz_robots/blob/4be09959c4c7c85872a34b8f0ee66516c1bb5a81/prbt_ikfast_manipulator_plugin/test/unit_tests/tst_prbt_ikfast_manipulator_plugin.test#L4
however this would mean cyclic dependency :-(

Currently
`colcon build` or `catkin_make_isolated --pkg prbt_ikfast_manipulator_plugin --catkin-make-args run_tests_prbt_ikfast_manipulator_plugin` (given prbt_moveit_config is not installed on the system)
results in something like

```
build/prbt_ikfast_manipulator_plugin/Testing/20200709-1902/Test.xml: 1 test, 0 errors, 1 failure, 0 skipped
- _ctest_prbt_ikfast_manipulator_plugin_rostest_test_unit_tests_tst_prbt_ikfast_manipulator_plugin.test
  <<< failure message
    ... logging to /root/.ros/log/rostest-06d8e700f584-23616.log
    Traceback (most recent call last):
      File "/opt/ros/melodic/share/rostest/cmake/../../../bin/rostest", line 36, in <module>
        rostestmain()
      File "/opt/ros/melodic/lib/python2.7/dist-packages/rostest/__init__.py", line 274, in rostestmain
        _main()
      File "/opt/ros/melodic/lib/python2.7/dist-packages/rostest/rostest_main.py", line 159, in rostestmain
        testCase = rostest.runner.createUnitTest(pkg, test_file, options.reuse_master, options.clear, options.results_base_dir)
      File "/opt/ros/melodic/lib/python2.7/dist-packages/rostest/runner.py", line 229, in createUnitTest
        config = roslaunch.parent.load_config_default([test_file], None)
      File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/config.py", line 461, in load_config_default
        loader.load(f, config, argv=args, verbose=verbose)
      File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 761, in load
        self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
      File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 733, in _load_launch
        self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
      File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 699, in _recurse_load
        val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
      File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 96, in call
        return f(*args, **kwds)
      File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 596, in _include_tag
        inc_filename = self.resolve_args(tag.attributes['file'].value, context)
      File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 189, in resolve_args
        return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
      File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 392, in resolve_args
        resolved = _resolve_args(resolved, context, resolve_anon, commands)
      File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 405, in _resolve_args
        resolved = commands[command](resolved, a, args, context)
      File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 166, in _find
        source_path_to_packages=source_path_to_packages)
      File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 212, in _find_executable
        full_path = _get_executable_path(rp.get_path(args[0]), path)
      File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 207, in get_path
        raise ResourceNotFound(name, ros_paths=self._ros_paths)
    rospkg.common.ResourceNotFound: prbt_moveit_config
```
see https://travis-ci.org/github/agutenkunst/pilz_robots/jobs/706632712

jschleicher, 2020-07-10:
What solution do you suggest?
* Move the ik test up into the *_moveit package?
* duplicate a minimal `test_context.launch` inside the *_ik package?
* ... other ideas?

agutenkunst, 2020-07-15:
Puuuh this is not easy. Somehow this seems fundamentally flawed since the *_ik packages is deeply bound to the robot_description it was generated from. (Sadly it does not generate some basic tests as well :-( ).

Possible solutions
1. Move the ik test up into the *_moveit package
  - Pro: Plugin is still tested locally.
  - Con: *_ik package is not self-contained. But as stated above it never really was...
  - Con: Coverage is calculated per package. Would need to exclude whole package.
2. Duplicate a minimal test_context.launch inside the *_ik package
  - Would not solve this problem since various files from prbt_moveit_config are needed
3. A "bit" of restructuring. See draft on https://github.com/PilzDE/pilz_robots/pull/440
  - Pro: Removes the cyclic dep by moving some files into prbt_support
  - Pro: Test could stay in place. Possible coverage calculation is not affected.
  - Con(?): pilz_trajectory_generation would need minor adaptions. But prbt_support is currently set as dep so no problem.
  - Con: Moves some config away from the default location set by MoveItSetupAssistant. Not sure if there is harm in this...
  - Note: 
       - https://github.com/PilzDE/pilz_robots/blob/melodic-devel/prbt_support/package.xml#L4-L5 could be extended to argue why these files could be moved here.

I would vote for https://github.com/PilzDE/pilz_robots/pull/440 but maybe there is an objection that I currently don't see.

@martiniil your opinion?

agutenkunst, 2020-07-15:
Found more (but maybe better for another PR):
https://github.com/PilzDE/pilz_robots/blob/12495bc53ac05538d3c92c8cf9d79efcb7c354cc/prbt_support/test/urdf_tests.test#L20

agutenkunst, 2020-07-28:
ping @martiniil

martiniil, 2020-07-29:
Hi, thanks for your effort. I lean towards 

> 1. Move the ik test up into the *_moveit package

The ik package was never included in the coverage so this is no problem. I think moving files away from the default location is misleading, whether you add an explanation or not.

Same for `urdf_test`. What do you think?

PR #450 (2020-08-07): Move ikfast plugin test to prbt_moveit_config
Fixes #439
closed 2020-08-07 by SansoneG