PilzDE/pilz_robots #444: PNOZ safety hardware uses wrong registers + missing modbus_write_api_pnoz.yaml

JonathanGruner, 2020-07-20:
### Commit
Current one

### Steps to reproduce
How can the issue/bug be reproduced?
1. Connect PNOZ
2. Download pilz_application_template
2. Change safety_hw parameter to "pnoz"

### Expected behavior
- Uses registers 512...515
- No further changes necessary

### Observed behavior
- Uses PSS4000 registers 970...980
- You have to make an empty file modbus_write_api_pnoz.yaml in the config folder because of errors when your program tries to read an non existend file

### Workaround
- Replace the modbus_<read/write>_client_pss4000.yaml with pnoz files to match with pnoz registers

- $ sudo touch /opt/ros/melodic/share/prbt_hardware_support/config/modbus_write_api_spec_pnoz.yaml

- $ sudo mv /opt/ros/melodic/share/prbt_hardware_support/config/modbus_write_api_spec_pnoz.yaml /opt/ros/melodic/share/prbt_hardware_support/config/modbus_write_api_spec_pss4000.yaml

- $ sudo mv /opt/ros/melodic/share/prbt_hardware_support/config/modbus_read_api_spec_pnoz.yaml /opt/ros/melodic/share/prbt_hardware_support/config/modbus_read_api_spec_pss4000.yaml

martiniil, 2020-07-23:
Thank you for reporting the issue.
The error with the missing file should disappear once you set `has_braketest_support:=false`. At least this is the intended setup for Pnoz.

martiniil, 2020-07-23:
@jschleicher This is already fixed here: #384

JonathanGruner, 2020-07-24:
> 
> 
> Thank you for reporting the issue.
> The error with the missing file should disappear once you set `has_braketest_support:=false`. At least this is the intended setup for Pnoz.

Thanks for the answer. I checked but the parameter is set to "false", therefore this can't be the reason. If you want I can send you my package.

martiniil, 2020-07-27:
> Thanks for the answer. I checked but the parameter is set to "false", therefore this can't be the reason. If you want I can send you my package.

Thanks Jonathan. Please post a log of the error. Let's see if this helps.

JonathanGruner, 2020-07-28:
[roslaunch-rospc-HP-ProBook-650-G5-10296.log](https://github.com/PilzDE/pilz_robots/files/4987227/roslaunch-rospc-HP-ProBook-650-G5-10296.log)

[my_application.launch.txt](https://github.com/PilzDE/pilz_robots/files/4987229/my_application.launch.txt)

martiniil, 2020-07-28:
Your log shows that the file `modbus_read_api_spec_pnoz.yaml` is missing. I guess this is due to your workaround, in which this file is renamed. Can you revert your workaround (completely!) and post the log of the original error?

JonathanGruner, 2020-07-28:
Ok you are right. Got confused with the errors and reseted the whole config folder into original. Than the error is gone.

martiniil, 2020-07-29:
Hi @JonathanGruner , this issue should be fixed right now on the branch `melodic-devel`. It would be nice if you could validate it.

JonathanGruner, 2020-08-20:
Hi,
it works with this branch melodic-devel. 
Tested it with prbt_default_application.
I will disassemble my testing cell now and close this request, thank you very much for your effort!

closed 2020-08-20 by JonathanGruner