PilzDE/pilz_robots #446: Report of performing acceptance-tests in test cell

martiniil, 2020-07-29:
### acceptance_test_speed_monitoring
Sometimes moving the robot with RViz fails with error:
```
[ERROR] [1596029290.842323725]: 
Invalid Trajectory: start point deviates from current robot state more than 0.01
joint 'prbt_joint_1': expected: -0.331473, current: -0.347024
```

### acceptance_test_execute_brake_test
Not yet supported by PSS4000 in test cell?
```
[ INFO] [1596029722.762158831]: Brake test result: 1
[DEBUG] [1596033365.438770251]: modbus_write_and_read_registers: writing from 982 2 registers                                      and reading from 970 1 registers
[ERROR] [1596029722.837969220]: Failed to write and read modbus registers: Slave device or server failure
[ERROR] [1596029722.838273941]: Modbus disconnect: Failed to write and read modbus registers: Slave device or server failure
[ INFO] [1596029722.839120850]: Operation Mode switch: 1 -> 0
[ERROR] [1596029722.839153563]: A disconnect from the modbus server happend.
```

### acceptance_test_stop1

controller is in holding mode at start -> Fix test description

### during acceptance_test_joint_limits

- controller is in holding mode at start -> Fix test description
- could not observe any log output indicating a joint limit violation
- trajectory is always successful, therefore the following assertion fails:
https://github.com/PilzDE/pilz_robots/blob/430fecffbf372ff574a776f5150988556d54df0a/prbt_support/test/acceptance_tests/acceptance_test_joint_limits.py#L220

martiniil, 2020-07-30:
Can be closed once #447 and #448 is merged.

jschleicher, 2020-10-09:
Thank you for the report. Closing as #447 and #448 are merged now.

closed 2020-10-09 by jschleicher