PilzDE/pilz_robots #474: miss OperationModes.h

lyh458, 2020-11-30:
### Commit
branch: melodic-devel

### Steps to reproduce
1. git clone https://github.com/PilzDE/pilz_robots.git
2. cd <workspace>
3. catkin_make
4. install the dependencies by command ``sudo apt install ros-melodic-pilz-<package>``, such as **pilz_msgs**, etc.

### Observed behavior

Could't found some headers when ``catkin_make``, even though I have installed the related packages, and could not foud in the installed packages' folder with command ``locate <header_name>``.
Logfiles below.

### Logfiles
```
In file included from /home/lyh/Codes/robot_ws/src/pilz_robots/prbt_hardware_support/src/libmodbus_client.cpp:25:0:
/home/lyh/Codes/robot_ws/src/pilz_robots/prbt_hardware_support/include/prbt_hardware_support/libmodbus_client.h:21:10: fatal error: modbus/modbus.h: No such file or directory
 #include <modbus/modbus.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
pilz_robots/prbt_hardware_support/CMakeFiles/pilz_modbus_client_node.dir/build.make:110: recipe for target 'pilz_robots/prbt_hardware_support/CMakeFiles/pilz_modbus_client_node.dir/src/libmodbus_client.cpp.o' failed
make[2]: *** [pilz_robots/prbt_hardware_support/CMakeFiles/pilz_modbus_client_node.dir/src/libmodbus_client.cpp.o] Error 1
CMakeFiles/Makefile2:19746: recipe for target 'pilz_robots/prbt_hardware_support/CMakeFiles/pilz_modbus_client_node.dir/all' failed
make[1]: *** [pilz_robots/prbt_hardware_support/CMakeFiles/pilz_modbus_client_node.dir/all] Error 2
[ 98%] Linking CXX shared library /home/lyh/Codes/robot_ws/devel/lib/libiiwa_hw.so
[ 98%] Built target iiwa_hw
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```

### Reference to test

martiniil, 2020-11-30:
1. Please clone https://github.com/PilzDE/pilz_common into your workspace. We moved some packages to this repository. However this change is not mapped to the apt-packages yet. You can skip this step once ros-melodic-pilz-msgs is at version 0.6.0

2. You are probably missing further dependencies. Rather than installing them by hand, consider using `rosdep` (http://wiki.ros.org/rosdep).

lyh458, 2020-11-30:
> 1. Please clone https://github.com/PilzDE/pilz_common into your workspace. We moved some packages to this repository. However this change is not mapped to the apt-packages yet. You can skip this [...]

Thanks your reply. 
I have solved the problem of missing **OperationModes.h, GetOperationMode.h** with your suggestion 1, but it still miss **modbus.h** . I tried to install the relative repo [libmodbus](https://github.com/PilzDE/libmodbus) in your repo list, another issue happened. Here is the log:

```
/mnt/WinDisk/Users/liyih/OneDrive - mail2.gdut.edu.cn/Ph.D/Codes/robot_ws/src/pilz_robots/prbt_hardware_support/src/libmodbus_client.cpp: In member function ‘virtual void prbt_hardware_support::LibModbusClient::setResponseTimeoutInMs(long unsigned int)’:
/mnt/WinDisk/Users/liyih/OneDrive - mail2.gdut.edu.cn/Ph.D/Codes/robot_ws/src/pilz_robots/prbt_hardware_support/src/libmodbus_client.cpp:74:51: error: invalid conversion from ‘timeval*’ to ‘uint32_t {aka unsigned int}’ [-fpermissive]
   modbus_set_response_timeout(modbus_connection_, &response_timeout);
                                                   ^~~~~~~~~~~~~~~~~
/mnt/WinDisk/Users/liyih/OneDrive - mail2.gdut.edu.cn/Ph.D/Codes/robot_ws/src/pilz_robots/prbt_hardware_support/src/libmodbus_client.cpp:74:68: error: too few arguments to function ‘int modbus_set_response_timeout(modbus_t*, uint32_t, uint32_t)’
   modbus_set_response_timeout(modbus_connection_, &response_timeout);
                                                                    ^
In file included from /mnt/WinDisk/Users/liyih/OneDrive - mail2.gdut.edu.cn/Ph.D/Codes/robot_ws/src/pilz_robots/prbt_hardware_support/include/prbt_hardware_support/libmodbus_client.h:21:0,
                 from /mnt/WinDisk/Users/liyih/OneDrive - mail2.gdut.edu.cn/Ph.D/Codes/robot_ws/src/pilz_robots/prbt_hardware_support/src/libmodbus_client.cpp:25:
/usr/local/include/modbus/modbus.h:186:16: note: declared here
 MODBUS_API int modbus_set_response_timeout(modbus_t *ctx, uint32_t to_sec, uint32_t to_usec);
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/mnt/WinDisk/Users/liyih/OneDrive - mail2.gdut.edu.cn/Ph.D/Codes/robot_ws/src/pilz_robots/prbt_hardware_support/src/libmodbus_client.cpp: In member function ‘virtual long unsigned int prbt_hardware_support::LibModbusClient::getResponseTimeoutInMs()’:
/mnt/WinDisk/Users/liyih/OneDrive - mail2.gdut.edu.cn/Ph.D/Codes/robot_ws/src/pilz_robots/prbt_hardware_support/src/libmodbus_client.cpp:80:68: error: cannot convert ‘timeval*’ to ‘uint32_t* {aka unsigned int*}’ for argument ‘2’ to ‘int modbus_get_response_timeout(modbus_t*, uint32_t*, uint32_t*)’
   modbus_get_response_timeout(modbus_connection_, &response_timeout);
                                                                    ^
pilz_robots/prbt_hardware_support/CMakeFiles/pilz_modbus_client_node.dir/build.make:110: recipe for target 'pilz_robots/prbt_hardware_support/CMakeFiles/pilz_modbus_client_node.dir/src/libmodbus_client.cpp.o' failed
make[2]: *** [pilz_robots/prbt_hardware_support/CMakeFiles/pilz_modbus_client_node.dir/src/libmodbus_client.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
```
BTW, I have also update the dependencies with the command ``rosdep install --from-paths src --ignore-src -r -y``.

lyh458, 2020-11-30:
> Thanks your reply.
> I have solved the problem of missing **OperationModes.h, GetOperationMode.h** with your suggestion 1, but it still miss **modbus.h** . I tried to install the relative repo [...]
I think this issue is similar to the closed issue #56, #143.

martiniil, 2020-11-30:
What is your operating system? The problem could be that you have an unsupported version of `libmodbus`. Currently we have libmodbus version 3.0.6-2 on Ubuntu 18.04.

lyh458, 2020-11-30:
> What is your operating system? The problem could be that you have an unsupported version of `libmodbus`. Currently we have libmodbus version 3.0.6-2 on Ubuntu 18.04.

Yes, the same as yours.

martiniil, 2020-11-30:
> > What is your operating system? The problem could be that you have an unsupported version of `libmodbus`. Currently we have libmodbus version 3.0.6-2 on Ubuntu 18.04.
> 
> Yes, the same as yours.

I have both `libmodbus5:amd64` and `libmodbus-dev` at version `3.0.6-2`. Can you check that?

lyh458, 2020-11-30:
> > > What is your operating system? The problem could be that you have an unsupported version of `libmodbus`. Currently we have libmodbus version 3.0.6-2 on Ubuntu 18.04.
> > 
> > 
> > Yes, the same as yours.
> 
> I have both `libmodbus5:amd64` and `libmodbus-dev` at version `3.0.6-2`. Can you check that?

Same too.
[image]

martiniil, 2020-11-30:
But this error
```
/libmodbus_client.cpp:74:68: error: too few arguments to function ‘int modbus_set_response_timeout(modbus_t*, uint32_t, uint32_t)
```
indicates that you have a higher version. Maybe you got your installations mixed up. Try to remove all libmodbus installations, run the `rosdep` command again and clean your workspace. Then it should hopefully work.

lyh458, 2020-12-01:
> But this error
>
> ```
> /libmodbus_client.cpp:74:68: error: too few arguments to function ‘int modbus_set_response_timeout(modbus_t*, uint32_t, uint32_t)
> ```
>
> indicates that you have a higher version. [...]

Problem solved. I uninstalled all libmodbus and installed again, then ``catkin_make``, it worked!
Thanks!

closed 2020-12-01 by lyh458