PilzDE/pilz_robots #488: PluginlibFactory: The plugin for class 'moveit_rviz_plugin/MotionPlanning' failed to load.

SantiagoCanete, 2021-07-19:
### Commit
Installed the library through 
`sudo apt install ros-melodic-pilz-robots`

### Setup
Using Ubuntu 20.04, ROS Noetic in WSL

### Tried
1. `sudo apt update`
2. `sudo apt-get dist-upgrade`
3. `sudo apt-get install ros-noetic-moveit-visual-tools`

### Expected behavior
Running the test launch:
`roslaunch prbt_moveit_config moveit_planning_execution.launch`

### Observed behavior
`[ERROR] PluginlibFactory: The plugin for class 'moveit_rviz_plugin/MotionPlanning' failed to load.  Error: Failed to load library /opt/ros/noetic/lib/libmoveit_motion_planning_rviz_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libfcl.so.0.6: cannot open shared object file: No such file or directory)`

martiniil, 2021-07-20:
I assume you installed `ros-noetic-pilz-robots`?

martiniil, 2021-07-20:
I tried it in a fresh container (docker) and could not observe an issue. Regarding WSL, I cannot help unfortunately.

SantiagoCanete, 2021-07-20:
Yeap, sorry I installed `ros-noetic-pilz-robots`. I edited the original issue. This seems to be related to WSL, unsetting the environment variable `export LIBGL_ALWAYS_INDIRECT=:` in .bashrc has solved the problem.

closed 2021-07-20 by SantiagoCanete